balena-os/balena-raspberrypi #420: RTL8188CUS devices do not work on RPi 2

majorz, 2019-11-12:
While a RTL8188CUS device would work on RPi 3, it does not show up at all on RPi 2. I verified with balenaOS 2.43.0+rev1 and Asus N10 Nano dongle. We have the same report for Adafruit Miniature Wifi (802.11B/G/N), which is on our supported list.

majorz, 2019-11-12:
Correction: it does not work on RPi 3 either. I got confused by the internal interface.

Looks like the appropriate driver is not included:
```
T:  Bus=01 Lev=02 Prnt=02 Port=03 Cnt=02 Dev#=  4 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0b05 ProdID=17ba Rev=02.00
S:  Manufacturer=Realtek
S:  Product=802.11n WLAN Adapter
S:  SerialNumber=00e04c000001
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#=0x0 Alt= 0 #EPs= 4 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
```

floion, 2019-11-12:
@shaunmulligan do we want this dongle supported on the rpi family?

majorz, 2019-11-13:
@floion it is one of the better cheap dongles I have here. Not the best stability, but definitely high on the list.

bbinet, 2019-11-13:
@majorz I would be interested to know the best ones on your list ;)

majorz, 2019-11-14:
@bbinet Atheros and MediaTek from my experience. My old TL-WN722N with Atheros chipset have never let me down.

The biggest issue with some WiFi chipsets is weak Linux driver support. Some companies may have nice hardware features, but maintain proprietary drivers and the in-kernel ones are not a priority. Some do not care much about Linux.

I have been planning for a while on making automated comprehensive tests and I hope to get started somewhere in the next weeks. We will publish the results once I am ready.

As a side note Techship has nice assortment of WiFi modules: https://techship.com/products/category/wifi-modules/. The USB ones are around $40, which is kinda expensive compared to the cheap dongles though.

bbinet, 2019-11-14:
@majorz thank you

janhapke, 2020-03-21:
The RTL8188CUS chipset is used in dongles by many different manufacturers and they were a very popular choice to get decent low cost WiFi into the Pi2. They were included in several Raspberry Pi Kits (Pi + Case + Wifi Dongle + PSU).

On Raspbian they work out of the box.

I was evaluating the Balena "Ecosystem" today with some spare Pi2 and it took me several hours to end up here realizing the reason nothing works is a missing wifi driver.

My guess is not many people will migrate their Raspberry 2 based fleets to Balena so it is debatable whether including support for this chipset makes sense. But it would be nice to at least include a note in the [Getting Started on the Raspberry Pi 2 Guide](https://www.balena.io/os/docs/raspberry-pi2/getting-started/) to check for Wifi Dongle Compatibility - it can save people like me who are just trying out Balena massive amounts of wasted time (and resulting frustration about Balena).

acostach, 2020-03-21:
Hi @janhapke , 

Thank you for sharing your experience with us!

Indeed, the Raspbian image will include more drivers, however, in order to keep a small footprint for the BalenaOS image some of these may not be included unless specifically requested. By not including all possible drivers we help keep data usage at a minimum for hostOS updates. You can find the list of supported adapters in our docs here: https://www.balena.io/docs/reference/hardware/wifi-dongles/

@garethtdavies maybe we can add a link to the supported adapters list in the getting started guide for the Pi2?

sharker92, 2020-03-31:
I spent some hours figuring out why my Wifi Dongle doesn't works. Found this really helpful, thanks.

garethtdavies, 2020-03-31:
Sorry for the delay in updating this but unlike the balena docs site (where this info is already [linked](https://www.balena.io/docs/learn/getting-started/raspberry-pi2/nodejs/)) the host OS site doesn't make use of partials so I can't (easily) add this on a per-device basis. That being said I've [created a PR](https://github.com/balena-io/resin-site/pull/974) for some generic text for supported WiFi adapters and modems.

sharker92, 2020-04-01:
> Sorry for the delay in updating this but unlike the balena docs site (where this info is already [linked](https://www.balena.io/docs/learn/getting-started/raspberry-pi2/nodejs/)) the host OS site [...]

Don't worry, I understand, thanks. By the way, the PR link is broken.

garethtdavies, 2020-04-01:
@sharker92 thanks, that PR is in a private repo so it probably appears broken for you. I’ve updated my earlier comment to reflect this.

srlowe, 2020-04-10:
I ran across this issue when I tried to use my old Pi2 (plus RTL8188CUS based dongle) in a project.  

@majorz I think it might be worth adding support for this chipset, as it was recommended quite a bit in forums when the Pi2 was popular.  There may still be a lot of hobbyist with this set up trying balena for the first time (and they might not come across the dongle support list, or know what the led errorcode means).

I noticed that when you try to provision a Pi2, you get a warning in the dashboard "Warning!The Raspberry Pi 2 is not capable of connecting to WiFi networks without an external adapter."  Seems like it would be useful to have a link to that supported dongle list there - I'll try to find who's responsible for that ui.

majorz, 2020-05-07:
The dongle apparently worked with earlier OS versions. Just checked with balenaOS 2.26.0+rev1:
```
T:  Bus=01 Lev=02 Prnt=02 Port=02 Cnt=02 Dev#=  4 Spd=480 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0b05 ProdID=17ba Rev=02.00
S:  Manufacturer=Realtek
S:  Product=802.11n WLAN Adapter
S:  SerialNumber=00e04c000001
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=500mA
I:  If#=0x0 Alt= 0 #EPs= 4 Cls=ff(vend.) Sub=ff Prot=ff Driver=rtl8192cu
```
This is the PR where it was disabled (in favor of another driver). I think we need to do more testing with both of those and determine which one is in a better shape and go with it.

spydon, 2020-08-19:
Was there ever any more testing done between the two drivers?