balena-os/balena-raspberrypi #424: unable to read resin-image-flasher

petaren, 2019-11-26:
I'm trying to deploy Balena to a RaspberryPi 3. I create the image on the website and download it, then using balenaEtcher on macOS I flash it to the SD-card. It verifies correctly in balenaEtcher. But when trying to boot it, it prints some data and then stops. After a while it reboots and the process starts over. 

RaspberryPi 3 B version 2
macOS: 10.15
balenaEtcher: 1.5.59
balenaOS: v2.38.0+rev1 (recommended)

The print from the rPi when balena boots is:

```
Net: No ethernet found.
switch to partitions #0, OK
mmc0 is current device
Scannin mmc 0:1...
Found U-Boot script /boot.scr
440 bytes read in 1ms (429.7KiB/s)
## Executing script at 02400000
Scanning mmc devices 0 1 2
** Unable to read file resin-image-flasher **
24 bytes read in 2ms (11.7KiB/s)
Card did not respond to voltage select!
** Bad device mmc 1 **
** Bad device mmc 2 **
Found resin image on mmc 0
Loading resinOS_uEnv.txt from mmc device 0 partition 1
** Unable to read file resinOS_uEnv.txt **
243230400 bytes read in 1103 ms (20.9 MiB/s)
** Flattened Device Tree blob at 2eff9700
    Booting using the fdt blob at 0x2eff9700
   reserving fdt memory region: addr=0 size=1000
   Using Device Tree in place at 000000002eff9700, end 000000002eff9700

Starting kernel ...
```

I suspected the problem might be the SD-card, so I bought a new one. Same error. I suspected it might be the rPi, so I bought a new one. Same problem.

ZubairLK, 2019-11-26:
The title of the issue mentions resin-image-flasher.. can you please elaborate what you mean by resin-image-flasher?

That is strange. can you try a dev image and see if you see anything more on the serial console?

petaren, 2019-11-26:
I'm mentioning `resin-image-flasher` because the log that the rPi prints mentions it. 

I am using a develop image.

petaren, 2019-11-26:
Also tried balenaEtcher 1.5.63 and using a Windows machine to flash with. Same results.

petaren, 2019-11-26:
I tried a different power adapter to the rPi and it booted now. No issues.

closed 2019-11-26 by petaren