balena-os/balena-raspberrypi #438: Boot halted when external device connected to UART continuously sends data

ffissore, 2019-12-23:
As reported [in the forums](https://forums.balena.io/t/multicontainer-uart-logging-program-keeps-rpi-from-booting-up/52541), balenaOS does not boot on a board which has an external device attached, continuously sending data to UART

This might be due to [this commit](https://github.com/balena-os/balena-raspberrypi/commit/2226245a633344a51b2696297d54f4baafcc2438) in which uboot flags for disabling boot interruption were removed in favour of a more generic approach at [meta-balena level](https://github.com/balena-os/meta-balena/pull/1349). However the meta-balena PR does not disable CTRL+C (`CONFIG_AUTOBOOT_KEYED_CTRLC`) so that an unlucky sequence sent by a device can still prevent a device from booting

floion, 2019-12-24:
Fixed by https://github.com/balena-os/meta-balena/pull/1785

floion, 2020-01-14:
This is now released

closed 2020-01-14 by floion