balena-os/balena-raspberrypi #444: [rpi3-64]: Running a full desktop in a container fails

acostach, 2020-01-08:
Container APP logs:

06.01.20 21:44:25 (+0200)  main  STARTING X
08.01.20 17:05:33 (+0200)  main  hostname: No address associated with hostname
08.01.20 17:05:34 (+0200)  main  xauth: (stdin):1:  bad display name "370a9e6:1" in "add" command
08.01.20 17:05:34 (+0200)  main  
08.01.20 17:05:34 (+0200)  main  
08.01.20 17:05:34 (+0200)  main  X.Org X Server 1.19.2
08.01.20 17:05:34 (+0200)  main  Release Date: 2017-03-02
08.01.20 17:05:34 (+0200)  main  X Protocol Version 11, Revision 0
08.01.20 17:05:34 (+0200)  main  Build Operating System: Linux 4.9.0-8-arm64 aarch64 Debian
08.01.20 17:05:34 (+0200)  main  Current Operating System: Linux 370a9e6 4.19.71 #1 SMP PREEMPT Mon Jan 6 19:29:45 UTC 2020 aarch64
08.01.20 17:05:34 (+0200)  main  Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=256M video=HDMI-A-1:1920x1080@60 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=null rootfstype=ext4 rootwait vt.global_cursor_default=0 root=PARTUUID=fe7a82b7-02 rootwait
08.01.20 17:05:34 (+0200)  main  Build Date: 03 November 2018  03:03:37AM
08.01.20 17:05:34 (+0200)  main  xorg-server 2:1.19.2-1+deb9u5 (https://www.debian.org/support) 
08.01.20 17:05:34 (+0200)  main  Current version of pixman: 0.34.0
08.01.20 17:05:34 (+0200)  main         Before reporting problems, check http://wiki.x.org
08.01.20 17:05:34 (+0200)  main         to make sure that you have the latest version.
08.01.20 17:05:34 (+0200)  main  Markers: (--) probed, (**) from config file, (==) default setting,
08.01.20 17:05:34 (+0200)  main         (++) from command line, (!!) notice, (II) informational,
08.01.20 17:05:34 (+0200)  main         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
08.01.20 17:05:34 (+0200)  main  (==) Log file: "/var/log/Xorg.0.log", Time: Wed Jan  8 15:05:34 2020
08.01.20 17:05:34 (+0200)  main  (==) Using system config directory "/usr/share/X11/xorg.conf.d"
08.01.20 17:05:35 (+0200)  main  MESA-LOADER: failed to retrieve device information
08.01.20 17:05:35 (+0200)  main  gbm: failed to open any driver (search paths /usr/lib/aarch64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
08.01.20 17:05:35 (+0200)  main  gbm: Last dlopen error: /usr/lib/dri/vc4_dri.so: cannot open shared object file: No such file or directory
08.01.20 17:05:35 (+0200)  main  failed to load driver: vc4
08.01.20 17:05:35 (+0200)  main  gbm: failed to open any driver (search paths /usr/lib/aarch64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
08.01.20 17:05:35 (+0200)  main  gbm: Last dlopen error: /usr/lib/dri/kms_swrast_dri.so: cannot open shared object file: No such file or directory
08.01.20 17:05:35 (+0200)  main  failed to load driver: kms_swrast
08.01.20 17:05:35 (+0200)  main  gbm: failed to open any driver (search paths /usr/lib/aarch64-linux-gnu/dri:${ORIGIN}/dri:/usr/lib/dri)
08.01.20 17:05:35 (+0200)  main  gbm: Last dlopen error: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory
08.01.20 17:05:35 (+0200)  main  failed to load swrast driver
08.01.20 17:05:35 (+0200)  main  couldn't get display device
08.01.20 17:05:36 (+0200)  main  
08.01.20 17:05:38 (+0200)  main  waiting for X server to begin accepting connections .
08.01.20 17:05:40 (+0200)  main  ..
08.01.20 17:05:42 (+0200)  main  ..
08.01.20 17:05:44 (+0200)  main  ..
08.01.20 17:05:46 (+0200)  main  ..

floion, 2020-01-08:
@acostach yeah, it should work every time after you reboot. Does it?

acostach, 2020-01-08:
@floion no, it only worked only after a while, after APP restarted by itself. It did NOT work after push or after reboot. This happens only on the 64bit version of the pi3
.

acostach, 2020-09-16:
Similar behavior on PI4 2020.07. Works fine after push and deploy, but after a reboot the App needs to be restarted for xfce to run

```
16.09.20 13:26:54 (+0300)  main  xinit: giving up
16.09.20 13:26:54 (+0300)  main  xinit: unable to connect to X server: Connection refused
16.09.20 13:26:54 (+0300)  main  xinit: server error
16.09.20 13:26:54 (+0300)  main  xauth: (argv):1:  bad display name "130c688:1" in "remove" command
16.09.20 13:26:55 (+0300)  main  STARTING X
16.09.20 13:26:56 (+0300)  main  hostname: No address associated with hostname
16.09.20 13:26:56 (+0300)  main  xauth: (stdin):1:  bad display name "130c688:1" in "add" command
16.09.20 13:26:56 (+0300)  main  
16.09.20 13:26:56 (+0300)  main  (EE) 
16.09.20 13:26:56 (+0300)  main  Fatal server error:
16.09.20 13:26:56 (+0300)  main  (EE) Server is already active for display 0
16.09.20 13:26:56 (+0300)  main         If this server is no longer running, remove /tmp/.X0-lock
16.09.20 13:26:56 (+0300)  main         and start again.
16.09.20 13:26:56 (+0300)  main  (EE) 
16.09.20 13:26:56 (+0300)  main  (EE) 
16.09.20 13:26:56 (+0300)  main  Please consult the The X.Org Foundation support 
16.09.20 13:26:56 (+0300)  main          at http://wiki.x.org
16.09.20 13:26:56 (+0300)  main   for help. 
16.09.20 13:26:56 (+0300)  main  (EE) 
```