balena-os/balena-raspberrypi #445: Latest version of balenaOS Raspberry Pi for Zero stuck at 2.32

imgrant, 2020-01-09:
I have some Raspberry Pi Zeros running Balena OS 2.32.0, which tell me that is the latest version if I try to update balenaOS from the dashboard (and, indeed, if I use the add device wizard, the image it downloads is 2.32).
However, one of the devices is running balenaOS 2.44.0, as at some point in the past (~last month or two), I _was_ able to update balenaOS via the dashboard.

What gives? Were the later releases pulled for the Zero? I note the changelog says the latest version is 2.46. When will be able to update Raspberry Pi Zeroes to later than 2.32?

floion, 2020-01-10:
Hi. Indeed, some release were pulled off. We are currently testing 2.46.1+rev1 and it will be available in a few days if all ok.
We advise in the meantime to not use in production the versions that have been removed.

floion, 2020-01-13:
Release `2.46.1+rev1` is now available in the dashboard.

closed 2020-01-13 by floion