balena-os/balena-raspberrypi #448: cant push cointainer in local mode with custom build

minusplusminus, 2020-01-10:
Hi,

I've added a recipe to the build, so that libqmi is added and initialized on boot. This should not have influence on other parts of the yocto build. 

using commit:
`v2.46.1+rev2`

OS:
Ubuntu 16.04.6

I executed:
`./balena-yocto-scripts/build/barys -m raspberrypi3 --shared-downloads  ~/downloads --bitbake-target resin-image -c -d`

Got some errors
```
ERROR: resin-image-1.0-r0 do_rootfs: /home/balena/balena-raspberrypi/build/tmp/deploy/images/raspberrypi3/u-boot.bin is an invalid path referenced in RESIN_BOOT_PARTITION_FILES.
ERROR: resin-image-1.0-r0 do_rootfs: Function failed: resin_boot_dirgen_and_deploy (log file is located at /home/balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/temp/log.do_rootfs.1128)
ERROR: Logfile of failure stored in: /home/balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/temp/log.do_rootfs.1128
ERROR: Task (/home/balena/balena-raspberrypi/build/../layers/meta-balena/meta-balena-common/recipes-core/images/resin-image.bb:do_rootfs) failed with exit code '1'

ERROR: resin-image-1.0-r0 do_rootfs: /home/balena/balena-raspberrypi/build/tmp/deploy/images/raspberrypi3/boot.scr is an invalid path referenced in RESIN_BOOT_PARTITION_FILES.
ERROR: resin-image-1.0-r0 do_rootfs: Function failed: resin_boot_dirgen_and_deploy (log file is located at /home/balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/temp/log.do_rootfs.2119)
ERROR: Logfile of failure stored in: /home/balena/balena-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/temp/log.do_rootfs.2119
ERROR: Task (/home/balena/balena-raspberrypi/build/../layers/meta-balena/meta-balena-common/recipes-core/images/resin-image.bb:do_rootfs) failed with exit code '1'
```

Got some errors, but after a few retries, it suddenly built the project. 

flashed the `resin-image-raspberrypi3-20200109185344.rootfs.resinos-img.xz`

It's running and I can ssh into the container. I see the  supervisor container `balena/armv7hf-supervisor:v10.6.27 ` running

```
Jan 09 18:55:26 neonlink resin-supervisor[1201]: Error response from daemon: No such container: resin_supervisor
Jan 09 18:55:26 neonlink resin-supervisor[1226]: active
Jan 09 18:55:28 neonlink resin-supervisor[1227]: Error: No such object: resin_supervisor
Jan 09 18:55:29 neonlink resin-supervisor[1227]: Error: No such container: resin_supervisor
Jan 09 18:55:33 neonlink resin-supervisor[1227]: Starting system message bus: dbus.
Jan 09 18:55:33 neonlink resin-supervisor[1227]:  * Starting Avahi mDNS/DNS-SD Daemon: avahi-daemon
Jan 09 18:55:33 neonlink resin-supervisor[1227]:    ...done.
Jan 10 15:16:11 neonlink resin-supervisor[1227]: [info]    Supervisor v10.6.27 starting up...
Jan 10 15:16:11 neonlink resin-supervisor[1227]: (node:1) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Starting event tracker
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Starting api binder
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Unmanaged mode is set, skipping API client initialization
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Starting logging infrastructure
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [event]   Event: Supervisor start {}
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Performing database cleanup for container log timestamps
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [info]    Starting up in unmanaged mode, activating local mode
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [info]    Switching logging backend to LocalLogBackend
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Storing engine snapshot in the database. Timestamp: 2020-01-10T15:16:12.227Z
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Handling of local mode switch is completed
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Connectivity check enabled: true
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   No API endpoint specified, skipping connectivity check
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Starting periodic check for IP addresses
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [info]    Reporting initial state, supervisor version and API info
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [info]    Attempting to load any preloaded applications
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [info]    Starting API server
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [info]    Applying target state
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [info]    Unmanaged mode is set, skipping API binder initialization
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Supervisor API listening on all interfaces
Jan 10 15:16:12 neonlink resin-supervisor[1227]: [debug]   Creating supervisor0 network
Jan 10 15:16:13 neonlink resin-supervisor[1227]: [debug]   Finished applying target state
Jan 10 15:16:13 neonlink resin-supervisor[1227]: [success] Device state apply success
Jan 10 15:18:00 neonlink resin-supervisor[1227]: [api]     GET /ping 200 - 11.008 ms
Jan 10 15:18:00 neonlink resin-supervisor[1227]: [api]     GET /v2/version 401 - 16.724 ms
Jan 10 15:21:03 neonlink resin-supervisor[1227]: [api]     GET /v1/healthy 200 - 11.669 ms
Jan 10 15:26:05 neonlink resin-supervisor[1227]: [api]     GET /v1/healthy 200 - 11.120 ms
Jan 10 15:26:11 neonlink resin-supervisor[1227]: [debug]   Attempting container log timestamp flush...
Jan 10 15:26:11 neonlink resin-supervisor[1227]: [debug]   Container log timestamp flush complete
Jan 10 15:31:06 neonlink resin-supervisor[1227]: [api]     GET /v1/healthy 200 - 11.362 ms
```

but when I try to push my container:
`The supervisor version on this remote device does not support multicontainer local mode. Please update your device to balenaOS v2.20.0 or greater from the dashboard.`

thanks in advance

floion, 2020-01-10:
@minusplusminus I will ask the supervisor maintainer on this one.
But what do you mean you added libqmi and initialized it? We already have libqmi in the rootfs. And how did you initialize a library?

minusplusminus, 2020-01-10:
oh didn't see libqmi.  I added an bash script that exectutes on boot: 
```

#!/bin/bash
echo "4" > /sys/class/gpio/export
sleep 0.1
echo "out" > /sys/class/gpio/gpio4/direction
echo "0" > /sys/class/gpio/gpio4/value
echo "6" > /sys/class/gpio/export
sleep 0.1
echo "out" > /sys/class/gpio/gpio6/direction
echo "0" > /sys/class/gpio/gpio6/value

set -x

PATH=/sbin:/bin:/usr/sbin:/usr/bin

qmicli -d /dev/cdc-wdm0 --dms-set-operating-mode='online'

qmicli -d /dev/cdc-wdm0 --dms-get-operating-mode
qmicli -d /dev/cdc-wdm0 --nas-get-signal-strength
qmicli -d /dev/cdc-wdm0 --nas-get-home-network
DEVICE=qmicli -d /dev/cdc-wdm0 -w
ip link set $DEVICE down
[ $? -ne 0 ] && echo "Failed to bring $DEVICE down" && exit 1
echo 'Y' | tee /sys/class/net/$DEVICE/qmi/raw_ip
[ $? -ne 0 ] && echo "Failed to change to raw-ip mode" && exit 1
ip link set $DEVICE up
[ $? -ne 0 ] && echo "Failed to bring $DEVICE up" && exit 1
qmicli -p -d /dev/cdc-wdm0 --device-open-net="net-raw-ip|net-no-qos-header" --wds-start-network="apn='portalmmm.nl',ip-type=4" --client-no-release-cid
[ $? -ne 0 ] && echo "Failed to connect to mobile network" && exit 1
udhcpc -i wwan0
[ $? -ne 0 ] && echo "Failed to get IP from provider" && exit 1
```

floion, 2020-01-10:
I see, we are testing the mPCIE SIM7600E on balena Fin for example and it passes testing with the OS image as is. Just so you know for the future, these are the modems and platforms we test them on: https://github.com/balena-os/meta-balena#modems

Can you paste the output of `cat /etc/os-release` please on the build where you get the supervisor error?

minusplusminus, 2020-01-10:
```
ID="balena-os"
NAME="balenaOS"
VERSION="2.44.0+rev4"
VERSION_ID="2.44.0+rev4"
PRETTY_NAME="balenaOS 2.44.0+rev4"
MACHINE="raspberrypi3"
VARIANT="Production"
VARIANT_ID="prod"
META_BALENA_VERSION="2.44.0"
RESIN_BOARD_REV="fe18823"
META_RESIN_REV="7fed82f3"
SLUG="raspberrypi3"

```

CameronDiver, 2020-01-13:
So this is a bit of bad error reporting from the CLI, where the 401 received from the version endpoint is interpreted as the device not having a new enough supervisor. The problem is actually because the supervisor is denying calls to it's endpoints (other than ping and healthcheck, which are not authenticated). To allow calls externally, the supervisor must be in local mode, but local mode is not possible with a production build.

I've created a CLI issue here: https://github.com/balena-io/balena-cli/issues/1565

floion, 2020-01-13:
Thanks Cameron.
So @minusplusminus you must use a development build and not a production one.
Can this issue be closed?

minusplusminus, 2020-01-13:
yes thanks

closed 2020-01-13 by acostach