balena-os/balena-raspberrypi #453: Missing `CONFIG_RPIVID_MEM` kernel config for Pi 4

richbayliss, 2020-02-03:


matohavo, 2020-02-05:
Where should this config with the [others](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/char/broadcom/Makefile) be added to rebuild the image with Broadcom driver?

floion, 2020-02-06:
Hi @matohavo what are you trying to achieve?

floion, 2020-02-06:
Fixed by https://github.com/balena-os/balena-raspberrypi/pull/457

matohavo, 2020-02-06:
_Disclaimer - I'm new to all of this, so I might be totally off._

Ultimately I'm trying to make hardware accelerated decoding of HEVC video work with BalenaOS on RPi4.

If I understand correctly, the [broadcom](https://github.com/raspberrypi/linux/tree/rpi-4.19.y/drivers/char/broadcom) driver needs to be used for that matter, because [ffmpeg patch](https://github.com/popcornmix/FFmpeg/commits/2711) requires video devices also from this driver:

```
/dev/rpivid-hevcmem
/dev/vc-mem
/dev/vcsm
/dev/vcsm-cma
/dev/vchiq
/dev/vcio
```

- Those devices are not present in current 64bit, so I'm not sure if [unavailability of vc4](https://github.com/agherzan/meta-raspberrypi/blob/master/docs/extra-build-config.md#vc4) on 64bit systems affect this somehow and if I have to use 32bit image. 
- I've tried to build 32bit image for RPi4 - the build succeeded after I added a coffee file, but it didn't boot. There was no coffee file for barys for 32bit RPi4, although the machine file is [there](https://github.com/agherzan/meta-raspberrypi/blob/master/conf/machine/raspberrypi4.conf) - is there a reason for that?
- guys from LibreELEC managed to get this working using V4L2, but they used newer kernel 5.4.16 - is it possible to build BalenaOS with new kernels?

Thanks

floion, 2020-02-09:
Hi, I can do a remote host OS update on a test device of yours if you wish and then you can check if the  CONFIG_RPIVID_MEM is all that's needed of if something else needs to be enabled.
We only ever tested the 64 bits OS and we have no plan going forward to add the 32 bits machine so no idea why it was not booting for you.
We will be switching to a new kernel once the BSP uses a newer kernel. I see that the newest BSP branch is `zeus` which has kernel 4.19.93 so 5.x is a long wait I would say.

matohavo, 2020-02-11:
Ok, thank you. I need these configs:
```
CONFIG_BRCM_CHAR_DRIVERS=y
CONFIG_BCM_VCIO=y
CONFIG_BCM_VC_SM=y
CONFIG_BCM2835_DEVGPIOMEM=y
CONFIG_RPIVID_MEM=y
```
Building vc_sm for 64bit is failing.

zub0r, 2020-04-01:
Thanks for suggested approach, it helped.
#close 453

floion, 2022-02-16:
Hi @zub0r is this still something you need?

commit 107ba77c (2020-02-07): linux-raspberrypi_%.bbappend: Enable CONFIG_RPIVID_MEM

This was requested by customer.

Fixes: #453
Changelog-entry: Enable the rpivid-mem kernel module
commit 76fbac68 (2020-04-24): linux-raspberrypi_%.bbappend: Enable CONFIG_RPIVID_MEM

This was requested by customer.

Fixes: #453
Changelog-entry: Enable the rpivid-mem kernel module