balena-os/balena-raspberrypi #460: 32bit image doesn't boot on RPi4

matohavo, 2020-02-07:
**Description**
I've built the BalenaOS image with [32 bit machine config](https://github.com/agherzan/meta-raspberrypi/blob/warrior/conf/machine/raspberrypi4.conf), but after flashing the image with etcher, RPi4 doesn't boot. (64 bit image works fine)

**Steps to reproduce the issue:**
1. I've cloned [balena-raspberrypi](https://github.com/balena-os/balena-raspberrypi) (used warrior branch of meta-raspberrypi)
2. I've created the raspberrypi4.coffee file from [raspberrypi4-64.coffee](https://github.com/balena-os/balena-raspberrypi/blob/master/raspberrypi4-64.coffee) and changed 
```
slug: 'raspberrypi4'
aliases: [ 'raspberrypi4' ]
arch: 'armv7hf'
machine: 'raspberrypi4'
```
3. Build with `./balena-yocto-scripts/build/barys -d -m raspberrypi4`

Am I missing something or it shouldn't work this way?

floion, 2020-02-09:
We have not tested the 32 bits image and we have no plans to support the 32 bits OS.
Did you connect a serial cable on the 6, 8 and 10 pins on the pinout header to check if there is any output?

floion, 2020-05-15:
Closing item as 32 bits is not something we support nor plan to support. 32 bits apps can be ran from the container on top of the 64 bits OS if need be.

closed 2020-05-15 by floion