balena-os/balena-raspberrypi #472: [RPI4] Rollback alt-boot fails between 2019.10.3 and 2020.04.0

acostach, 2020-04-07:
MMC:   emmc2@7e340000: 0, mmcnr@7e300000: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:  serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
switch to partitions #0, OK
mmc0 is current devic
Scanning mmc 0:1...
Found U-Boot script /boot.scr
440 bytes read in 6 ms (71.3 KiB/s)
## Executing script at 02400000
Scanning mmc usb devics 0 1 2
24 bytes read in 11 ms (2 KiB/s)
Found resin image on mmc 0
Loading resinOS_uEnv.txt from mmc device 0 partition 1
38 bytes red in 14 ms (2 KiB/s)
Import resinOS_uEnv.txt in environment
13 bytes read in 14 ms (0 Bytes/s)ice 0 partition 1
Import bootcount.env in environment
bootcount=5 now
13 bytes written
bootcount=5 written to mmc 0: bootcount.env
WARNING! BOOTLIMIT EXCEEDED. SWITCHING TO PREVIOUS ROOT
WARNING! was: resin_root_part=3
WARNING!now: resin_root_part=2
23187968 bytes read in 1810 ms (12.2 MiB/s)
## Flattened Device Tree blob at 2eff5c00
   Bootingusing the fdt blob at 0x2eff5c00
   reserving fdt memory region: addr=0 size=1000
   Using Device Tree in place t 000000002eff5c00, end 000000002f002f2b

Starting kernel ...

MMC:   emmc2@7e340000: 0, mmcnr@7e300000: 1
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:  serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
switch to partitions #0, OK
mmc0 is current devic
Scanning mmc 0:1...
Found U-Boot script /boot.scr
440 bytes read in 6 ms (71.3 KiB/s)
## Executing script at 02400000
Scanning mmc usb devics 0 1 2
24 bytes read in 11 ms (2 KiB/s)
Found resin image on mmc 0
Loading resinOS_uEnv.txt from mmc device 0 partition 1
38 bytes red in 14 ms (2 KiB/s)
Import resinOS_uEnv.txt in environment
13 bytes read in 14 ms (0 Bytes/s)ice 0 partition 1
Import bootcount.env in environment
bootcount=6 now
13 bytes written
bootcount=6 written to mmc 0: bootcount.env
WARNING! BOOTLIMIT EXCEEDED. SWITCHING TO PREVIOUS ROOT
WARNING! was: resin_root_part=3
WARNING!now: resin_root_part=2
23187968 bytes read in 1810 ms (12.2 MiB/s)
## Flattened Device Tree blob at 2eff5c00
   Bootingusing the fdt blob at 0x2eff5c00
   reserving fdt memory region: addr=0 size=1000
   Using Device Tree in place t 000000002eff5c00, end 000000002f002f2b

Starting kernel ...

And this is where kernel hangs forever. Happens with both dev and prod images.

acostach, 2020-04-07:
On my side this appears to happen due to booting the old kernel with the new dtbs or overlays, when switching back to the old OS.

After copying the dtb and overlays from the resin-rootA to resin-boot partition, booting the old OS kernel appears to work.

floion, 2020-07-27:
This DID NOT happen when hupping from `v2.51.1+rev1` and `v2.53.9+rev1` on rpi3 32 bits (there was a kernel bump from version `4.19.75` to `4.19.118` to in release `v2.51.1+rev2`).

acostach, 2020-09-16:
Fails on 2020.04 - to 2020.07

acostach, 2021-02-18:
2020.07 - 2021.01.0, happens when rolling back to 2020.07, the new kernel 5.4 dtb and fw from the boot partitions are used with the old 4.19 kernel, and no /dev/mmcblk* entries are not present in initramfs. Copying the 4.19 dtb and firmware in the boot partition solves the issue. The dtb is loaded by the firmware from the boot partition, and these contents cannot be altered until rolling back into a working OS.