balena-os/balena-raspberrypi #485: vcgencmd missing

chrisys, 2020-05-12:
On balenaOS for the Pi 4, `vcgencmd` is present, but on versions for other devices, such as 64-bit Pi 3, Pi Zero, Fin, this binary is missing.

floion, 2020-05-15:
Hi @chrisys what is the usecase for needing this utility?

saintaardvark, 2020-05-28:
For myself, I'd love to have this available to use in https://github.com/balena-io/device-diagnostics to  detect CPU throttling more reliably on RPis.   

For context, see https://github.com/balenalabs/rosetta-at-home/issues/47 and https://github.com/balena-io/device-diagnostics/issues/183.

xZero707, 2022-07-24:
> Hi @chrisys what is the usecase for needing this utility?

I think this is really unnecessary question. On Raspberry Pi, `vcgencmd` is kind of mandatory, especially for developers.