balena-os/balena-raspberrypi #486: 64bit balenaOS not booting on Raspberry Pi CM3+ in Waveshare carrier board

HakanL, 2020-05-14:
I'm trying to get 64-bit balenaOS to run on a RPi CM3+ (32GB) compute module sitting in a WaveShare carrier board.
Works:
32-bit balenaOS
64-bit Ubuntu Server (20.04 LTS)

Doesn't work:
64-bit balenaOS, version 2.47.0+rev1

The image works fine on a regular RPi3, so it's most likely an issue with the WaveShare board. The manufacturer does not provide any Yocto files unfortunately. When starting up there's no text on the monitor, but it turns on. Also no network activity is detected. I tried a serial connection to the board but nothing shows up (but I'm not sure I'm doing that part right).
Also see this discussion on the forum: https://forums.balena.io/t/waveshare-compute-module-poe-with-raspberry-pi-cm3-32gb/110477

floion, 2020-05-15:
Hi @HakanL can you try out the `2.38.0` 64 bits release first as a quick test?
Can you in parallel set-up a serial connection on a .dev 32 bits OS version and check that you are doing all ok there? If you're setting it up correctly, you should see the bootlog and at the end of the boot process you will be presented with the login prompt. If you then manage to set-up the serial console, then use a .dev 64 bits OS release and check if there's any output.

HakanL, 2020-05-19:
Version 2.38.0+rev1 (from 2019) only worked in 32-bit, not 64-bit on the RPi CM3+ module. I'm working on getting serial debug info to work.

HakanL, 2020-05-19:
I tried version 2.47rev1 32-bit on the RPi CM3 on the WaveShare carrier with serial connected. I can see the debug messages and everything. But when I test the 64-bit image (same version) I don't get anything on the serial port at all.

ryanteck, 2020-12-16:
This is also affecting us. Posted on the forum at https://forums.balena.io/t/balena-os-64-bit-does-not-work-on-compute-module-3/223502

We've got a Compute Module board that we'd like to use Balena with, however as the software is 64 Bit need to use that instead of the usual armv7hf envrionment.

It seems that RPi CM3 64 Bit hasn't been updated since January? 

It looks to me as if because it's not been updated that it's not had the new device tree files added for 64 Bit CM3/+. I have it semi working by copying in the device tree files to a boot partition but it's not stable but enough to confirm that's whats missing.

32 Bit works fine.

shawaj, 2020-12-17:
I have reproduced this issue on a Fin v1.1.1 with CM3+ lite as well.

Had to copy across the balena-fin.dtb and the bcm2710-rpi-cm3.dtb and add `dtoverlay=balena-fin` to the config.txt file

shawaj, 2020-12-17:
PRs to fix:
https://github.com/agherzan/meta-raspberrypi/pull/783
https://github.com/balena-os/balena-raspberrypi/pull/580

For CM3 this should work with these updates. If wanting to use raspberrypi3-64 with a balenaFin you will still need to add dtoverlay=balena-fin to the config.txt

I guess the only way around this would be to add a fincm3-64 device but not sure if there is much point.

shawaj, 2021-02-07:
@HakanL this is fixed in #594 and also see https://github.com/balena-os/balena-raspberrypi/issues/595#issuecomment-774708770

closed 2023-05-03 by HakanL