balena-os/balena-raspberrypi #524: [Rpi4 8GB] Reported u-boot failure, only with the 8G version

acostach, 2020-08-13:
As reported by @mtoman , the 8GB variant of the board [resets](https://www.flowdock.com/app/rulemotion/r-resinos/threads/M83xUSa_TmdkMf9QTSySSMQ3e2D) in u-boot when trying to initialize usb. This does not happen with the 4G version.

With the latest Ubuntu server for the Pi4 64bit this issue is also present.

Will need to reproduce and investigate this when the 8GB unit arrives.

jimsynz, 2020-08-13:
I have two of these at home. I'm happy to help any way I can.

dstewen, 2020-08-20:
I also have 4 x Pi4 8GB that I have been trying to get working. Happy to test any possible fixes.

vicgal, 2020-08-24:
Fixed in https://github.com/balena-os/balena-raspberrypi/commit/4df2f4ed46cffe830dbe6e677b2abc20cc25f96c, but not yet PRed.

For the build to succeed it needs this patch applied to meta-balena:
```
diff --git a/meta-balena-common/recipes-devtools/kernel-modules-headers/files/gen_mod_headers b/meta-balena-common/recipes-devtools/kernel-modules-headers/files/gen_mod_headers
index 9d564a6a..baee30d0 100755
--- a/meta-balena-common/recipes-devtools/kernel-modules-headers/files/gen_mod_headers
+++ b/meta-balena-common/recipes-devtools/kernel-modules-headers/files/gen_mod_headers
@@ -207,6 +207,14 @@ if [ "$karch" = "arm64" ]; then
        fi
 fi
 
+
+if [ "$karch" = "arm64" ]; then
+       if [ -f "lib/vdso/Makefile" ]; then
+               copy_mkdir "$target_dir" "./lib/vdso/Makefile"
+       fi
+fi
+
+
```

floion, 2020-09-07:
This is fixed by a combo of PRs:
https://github.com/balena-os/balena-raspberrypi/pull/528
https://github.com/balena-os/balena-raspberrypi/pull/529
https://github.com/balena-os/balena-raspberrypi/pull/530

The release `v2.56.0+rev1` should have this fixed

closed 2020-09-07 by floion