balena-os/balena-raspberrypi #526: RPI4 USB Boot : i2c-gpio overlay (OS 2.53.12)

apage224, 2020-08-14:
I've successfully been able to boot a PI 4 via USB. Only issue I am having is that enabling i2c-gpio overlay prevents the Pi from booting. The Pi will fail if I add the following to config.txt:
`dtoverlay=i2c-gpio,bus=3,i2c_gpio_delay_us=1,i2c_gpio_sda=26,i2c_gpio_scl=16`
If I remove this then it will boot successfully again. I know there are additional i2c busses but unfortunately board I am interfacing was done before the PI 4 so pins don't align. 

As a workaround I have attempted to dynamically add overlays via dtoverlay command from Host OS, however, this also gives me issues regardless of overlay I try...

```
root@e5814b7:~# dtoverlay -v -d /resin-boot/overlays i2c-gpio bus=3 i2c_gpio_delay_us=1 i2c_gpio_sda=8 i2c_gpio_scl=7
DTOVERLAY[debug]: Found override bus
DTOVERLAY[debug]:   override bus: cell target reg @ offset 0 (size 4)
DTOVERLAY[debug]: Found override i2c_gpio_delay_us
DTOVERLAY[debug]:   override i2c_gpio_delay_us: cell target i2c-gpio,delay-us @ offset 0 (size 4)
DTOVERLAY[debug]: Found override i2c_gpio_sda
DTOVERLAY[debug]:   override i2c_gpio_sda: cell target gpios @ offset 4 (size 4)
DTOVERLAY[debug]: Found override i2c_gpio_scl
DTOVERLAY[debug]:   override i2c_gpio_scl: cell target gpios @ offset 16 (size 4)
DTOVERLAY[debug]: delete_node(/__symbols__)
DTOVERLAY[debug]: Wrote 1023 bytes to '/tmp/.dtoverlays/0_i2c-gpio.dtbo'
DTOVERLAY[debug]: Wrote 1080 bytes to '/sys/kernel/config/device-tree/overlays/0_i2c-gpio/dtbo'
* Failed to apply overlay '0_i2c-gpio' (kernel) 
```

floion, 2020-09-09:
Hi, can you check with release `2.56.0+rev1`?

adamtec, 2020-10-08:
Sorry for the delay. Yes with 2.56.0+rev1 it is working as expected. I've tested on both fresh install as well updating existing OS on RPI4.

acostach, 2020-10-08:
Thank you for confirming, closing ticket as 2.56.0+rev1 is available in production.

closed 2020-10-08 by acostach