balena-os/balena-raspberrypi #546: GPIO pin 8 is busy on a Pi4 after an update from 2.51.1+rev1 to 2.56.0+rev1

mtoman, 2020-10-09:
With `2.51.1+rev1` GPIO pin 8 was available for programming by default but after a hup to `2.56.0+rev1` the pin is busy by default. Investigate why this happens. There has been a kernel bump, might have pulled some extra bits.

There is a reproducer app here: https://github.com/hpgmiskin/balena-gpio-debug

hpgmiskin, 2020-10-09:
I think this issue also effects GPIO pin 7

acostach, 2020-10-12:
Hello Henry,

The Pi4 kernel was updated between BalenaOS version 2.51.1 and 2.56, and the new kernel 5.4 no longer allows for gpios used by spi to be exported. Gpios 7 and 8 are defined in the device tree as chip selects for spi, which is enabled by default.

If you need these pins, the first solution would be to disable spi from the dashboard: Device Configuration -> Define DT parameters -> Remove "spi=on".

Second option would be to reassign the CS pins to some other GPIOs you don't need. In the dashboard, edit "Device configuration -> Define DT overlays" as described here: https://github.com/raspberrypi/linux/issues/3749#issuecomment-663401962

acostach, 2020-10-19:
Hi @hpgmiskin , do any of these solutions work for you?

hpgmiskin, 2020-10-21:
Hi @acostach sorry for the delay in response. Disabling spi worked perfectly and we no longer experience this issue on the latest BalenaOS. Thanks for your help

acostach, 2020-10-21:
You are welcome, thanks for confirming.

closed 2020-10-21 by acostach