balena-os/balena-raspberrypi #551: Pack the generated defconfig along prebuilt modules

acostach, 2020-10-20:
We could include the generated defconfig along the prebuilts for the wlan drivers in https://github.com/balena-os/balena-raspberrypi/tree/master/layers/meta-balena-raspberrypi/recipes-kernel/sd8887-mrvl/prebuilts , so that it can be re-used for custom OS builds. A similar defconfig needs to be used for the kernel to load these pre-built drivers, otherwise loading will fail due to mismatching symbol versions.

crazystick, 2021-02-18:
Since the switch to Dunfell, the default kernel version moved to 5.4 but the sd8887 prebuilts are still for 4.19 

Would be good to get new versions and also have this issue solved, as they will be unusable without the matching config

alexgg, 2021-03-04:
The defconfig is already available.

closed 2021-03-04 by alexgg