balena-os/balena-raspberrypi #561: RPi4 fails to boot with USB keyboard connected

markcorbinuk, 2020-11-10:
Raspberry Pi 4 is not booting when a USB keyboard is connected under balenaOS 2.56.0 and 2.58.6 (works with 2.51.1).

It appears that the later versions of OS use a different version of U-Boot which scans for USB devices at boot. The error message reported is as follows:
[image]

xginn8, 2020-11-10:
To clarify: the issue seems specific to the Apple MB110D/B with built-in USB Hub keyboard, not generically applicable.

alanb128, 2020-11-10:
Just capturing this item from Alexandru in the thread: If someone has a similar keyboard on which we could repro the issue we could try the last patch of this series, which is still pending: http://u-boot.10912.n7.nabble.com/RFC-PATCH-v2-0-5-Improve-USB-Keyboard-support-for-rpi3-rpi4-td418314.html#a418316

bblanke, 2020-12-10:
I've had the same issue with a USB touchscreen attached.

acostach, 2020-12-18:
@bblanke you can try with 2.65.0 for the Pi4. Was seeing the same issue with an arduino. Might take a little longer to boot though, till the usb timeouts expire in u-boot.

closed 2020-12-18 by acostach