balena-os/balena-raspberrypi #567: Add support for RPi400

mtoman, 2020-11-20:
RPi400 has its own device tree and does not work out of box with the Pi4 image. This is the u-boot log:
```
U-Boo 2020.07 (Oct 21 2020 - 11:03:27 +0000)

DRAM:  3.9 GiB
RPI: Board rev 0x13 outside known range
RPI Unknown model (0xc03130)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
Loading Environment from FAT... ** No partition table - mmc 0 **
In:    serial
Out:   serial
Err:   seral
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: probe failed, error -110
No working contollers found
switch to partitions #0, OK
mmc0 is current device
** No partition table - mmc 0 **
Card did not respond to voltage select!
                                       starting USB...
Bus xhci_pci: probe failed, error -110
No working controllers found
USB is stopped. Please issue 'usb tart' first.
starting USB...
Bus xhci_pci: probe failed, error -110
No working controllers found
ethernet@7d580000 Waitingfor PHY auto negotiation to complete.... done
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
BOOTP broadcast 6
BOOTP broadcast 8
BOOTP broadcast 9
BOOTP broadcast 10
BOOTP broadcast 11
BOOTP broadcast 12
BOOTP broadcast 13
BOOTP broadcast 14
BOOTP broadcast 15
BOOTP broadcast 16
BOOTP broadcast 17

Retry time xceeded; starting again
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrievin file: pxelinux.cfg/01-dc-a6-32-d3-ec-0b
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retieving file: pxelinux.cfg/00000000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrievin file: pxelinux.cfg/0000000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file:pxelinux.cfg/000000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinx.cfg/00000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
*** ERROR: `serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
*** ERROR: serverip' not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
*** ERROR: `serverip' ot set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
*** ERROR: `serverip' not set
missig environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-bcm283x-rpi
*** ERROR: `serverip' not et
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-bcm283x
*** ERROR: `serverip'not set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
*** ERROR: `serverip' no set
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
*** ERROR: `serverip' not set
onfig file not found
starting USB...
Bus xhci_pci:
```

acostach, 2020-11-25:
We'll also have to bring in the firmware for wifi: 
`brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43456-sdio.bin failed with error -2`

acostach, 2020-11-30:
Pi400 board did not arrive yet for our team, we'll resume investigation for adding support for it after we receive it.

shawaj, 2020-12-08:
@acostach I added the brcmfmac43456-sdio firmware and clm_blob from [here](https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm) to /lib/firmware/brcm and the WiFi started working 👍 

Patch here https://github.com/agherzan/meta-raspberrypi/pull/777

Hope that's the right place?

acostach, 2020-12-08:
Yes, that sounds fine, thanks @shawaj

shawaj, 2020-12-08:
I guess this goes without saying @acostach but I also had to bring in the new driver and compile it to add the definitions for the BCM43456 in the [sdio.c file](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/net/wireless/broadcom/brcm80211/brcmfmac/sdio.c)

acostach, 2020-12-08:
good point @shawaj , I was looking at bumping the kernel from 5.4.58 to 5.4.77 for the pi4 in https://github.com/balena-os/balena-raspberrypi/pull/566 to include the dtb for this board, I'll have to take a look if this tag includes the right driver. Board didn't arrive yet though. 

Note to self: I'll need to check if this [change](https://github.com/balena-os/balena-raspberrypi/commit/4bfe6d3dba00b072df5d02ddea7ce8a571baad70) needs to be ported to #566 too.

shawaj, 2020-12-08:
@acostach looks like it was added in April 2019 - https://github.com/raspberrypi/linux/commit/e3062e05e1cfe378bb9b3fa0bef46711372bcf13

Although I think it was first included in a release here - https://github.com/raspberrypi/linux/releases/tag/raspberrypi-kernel_1.20200527-1

Also see here - https://github.com/raspberrypi/linux/releases/tag/rpi53_2019019

So I guess it should be in 5.4.58 already....I am on balenaOS ESR v2020.07.02 which seems to be on kernel 4.19.75 so I guess that is why it is outdated. Will update to v2.58.3+rev1 and I think that should have the new brcmfmac.ko file already

shawaj, 2020-12-17:
Linking to #577 which has been merged

acostach, 2020-12-17:
Short update @alanb128  @mtoman @shawaj the device is now available for testing in the staging dashboard

shawaj, 2020-12-17:
@acostach sweet will test this eve - is staging available for customers as well? I think so but just wanted to check before suggesting it to anyone.

acostach, 2020-12-17:
@shawaj yes, it is available for everybody

shawaj, 2020-12-17:
@acostach working nicely for me on Pi400. Got a lora gateway running on it haha.

Does this fix bring in the WiFi driver for the CM4 to the Pi 4 image as well by any chance?

shawaj, 2020-12-20:
Closing as this is now available in [main dashboard](https://dashboard.balena-cloud.com/)

closed 2020-12-20 by shawaj