balena-os/balena-raspberrypi #570: Infinite reboot loop when USB connected Raspberry Pi 4

santicomp2014, 2020-11-29:
I have an issue with Balena OS, I'm using an Arduino Mega 2560 connected via USB directly to the Raspberry PI 4.
If I boot the Rpi with the USB plugged in, it reboots and does not start (Infinite restart loop), but if I unplug it and wait for the OS to boot, I can connect via serial port from the Balena dashboard.

This is really odd, I found some other people with similar [issues](https://www.raspberrypi.org/forums/viewtopic.php?t=266853)
Raspian with the same configuration boots correctly.

Is there a way to delay the init of the USB, I don't think it's underpowered.

How can I help with this bug?, or is there any other solution I could use?.
Thanks

acostach, 2020-11-30:
Hi @santicomp2014, if you use a development image and connect a serial cable to the Pi, do you see something like:

```
USB XHCI 1.00
scanning bus xhci_pci for devices... WARN halted endpoint, queueing URB anyway.
Unexpected XHCI event TRB, skipping... (3e164300 00000000 1300)
BUG!at drivers/usb/host/xhci-ring.c:498/abort_td()!

resetting ...
```

or something else?

santicomp2014, 2020-11-30:
> Hi @santicomp2014, if you use a development image and connect a serial cable to the Pi, do you see something like:
>
> ```
> USB XHCI 1.00
> scanning bus xhci_pci for devices... WARN halted endpoint, [...]

Yes, It's exactly that error.

```
U-Boot 2020.07 (Oct 21 2020 - 11:03:27 +0000)

DRAM:  7.7 GiB
RPI 4 Model B (0xd03114)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:   erial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPort 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... WARN halted endpoint, queueing URB anyway
Unexpected XHCI event TRB, skipping... (2eb543c0 00000000 13000000 02008401)
BUG at drivers/usb/host/xhci-ring.c:498/abort_td()!
                                                   BUG!
```

santicomp2014, 2020-11-30:
Thanks for the fix.

closed 2020-11-30 by santicomp2014

acostach, 2020-12-01:
@santicomp2014 let me know if the testing image 2.60.1+rev6 from https://dashboard.balena-staging.com works for you. It will take a bit longer for the image to boot when the arduino is connected to the usb, but should boot eventually.

santicomp2014, 2020-12-01:
> @santicomp2014 let me know if the testing image 2.60.1+rev6 from https://dashboard.balena-staging.com works for you. It will take a bit longer for the image to boot when the arduino is connected to the usb, but should boot eventually.

@acostach It took a bit longer, but it works.

Thanks for the help, when does this version get promoted into `dashboard.balena-cloud.com` is there a release cycle?.
I have 50 raspberry pi's that will be using this image, very happy to get this running.

acostach, 2020-12-02:
You are welcome. Probably in a week or two..

acostach, 2020-12-18:
@santicomp2014 2.65.0 with the patch included is now available in the dashboard for the Pi4.