balena-os/balena-raspberrypi #572: Raspberry Pi 4 /w official touchscreen udev race condition

haata, 2020-12-09:
With the official Raspberry Pi Touchscreen on the Raspberry Pi 4 I've been having issues getting the touchscreen to show up in x11 applications (as well as libinput).

Turns out the issue was udev didn't see the device initially and there was no hotplug event.
/dev/input/event0 did exist, but udev didn't know about the device.

Current workaround is to call (from balenaOS, doesn't work from inside a container):
```
udevadm trigger
```

I must have spent a week trying to track this down as it wasn't obvious at all...

marcosnils, 2020-12-13:
@haata just came across this issue by accident while building balena for a pi4 project that I have which also involves a touch screen. Do you think something can be done to prevent having to do this manually in the OS? It would be awesome to fix this without having to manually trigger udev rules.

haata, 2020-12-13:
I created this issue which located the (likely) root problem: https://github.com/balena-io-library/base-images/issues/670

https://github.com/balena-io-library/base-images/blob/91228b8f7785c025ac662ff38d42b119364ccdfd/balena-base-images/aarch64/debian/bullseye/run/entry.sh#L47
This line (`udevadm trigger &> /dev/null`) needs to be changed to:

```bash
chroot /proc/1/root /bin/bash -c "udevadm trigger"
```

I haven't found any other way to do this so far.