balena-os/balena-raspberrypi #576: Update to yocto 3.x?

marcosnils, 2020-12-13:
While trying to build balenaOS for rpi4 I've noticed that it was still using Yocto 2.7 which is considered obsolote. I thought that changing the octo version in the barys definitions (https://github.com/balena-os/balena-raspberrypi/blob/master/raspberrypi4-64.coffee#L25) was going to force the build system to bump bit after that I've noticed that Poky submodule is pinned to warrior version. 

Are there any plans to bump to dunfell any time soon?

acostach, 2020-12-14:
Hi @marcosnils , yes, Dunfell support was recently merged in https://github.com/balena-os/meta-balena/pull/2045 and the plan is to bump this device repository to Dunfell too.

marcosnils, 2020-12-14:
Awesome! I was trying to build balena-os while using ubuntu 20.04 as a host OS and I'm getting a glibc version error in the do_image_hostapp_ext4 task almost at the very end of the compilation that I'm sure is related to this. I'll try to figure out how to overcome this in the mean time.

floion, 2021-02-26:
We are now using dunfell for this repo so closing this item

closed 2021-02-26 by floion