balena-os/balena-raspberrypi #581: Upstreaming of balena-fin-overlay.dts changes?

shawaj, 2020-12-17:
I was just wondering if there was any particular reason why these patches had not been upstreamed:

https://github.com/balena-os/balena-raspberrypi/blob/master/layers/meta-balena-raspberrypi/recipes-kernel/linux/linux-raspberrypi/0004-mmc-pwrseq-Repurpose-for-Marvell-SD8777.patch
https://github.com/balena-os/balena-raspberrypi/blob/master/layers/meta-balena-raspberrypi/recipes-kernel/linux/linux-raspberrypi/0005-balena-fin-wifi-sta-uap-mode.patch

I am happy to submit some PRs in [raspberrypi/linux](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/overlays/balena-fin-overlay.dts) but just wanted to see if there was a reason not to first?

majorz, 2020-12-17:
One of those patches needs to be completely reworked and submitted to the mainline kernel instead. After it lands to stable then it can be merged with the raspberrypi kernel. We discussed those patches @alexgg recently in relation to his work with the Rapbian Fin support. I think we can discuss this further on FD as it will be a more appropriate place.

shawaj, 2020-12-17:
@majorz ok great, makes sense 👍