balena-os/balena-raspberrypi #584: Raspberry Pi CM4 Support

shawaj, 2020-12-21:
As per [this forum post](https://forums.balena.io/t/balenaos-on-raspberry-pi-400/211590/62) the CM4 is working nicely with balenaCloud but only using the Pi400 image.

Perhaps we need to update the Pi 4 image to include the necessary stuff for CM4 support.

Or maybe a new image is better. Not sure.

In any case just creating this issue for tracking purposes for anyone interested.

immiz, 2021-02-01:
is this issue already planned somewhere on your roadmap?

shawaj, 2021-02-02:
@immiz yes very much on our roadmap. We rely on the meta-raspberrypi repo here https://github.com/agherzan/meta-raspberrypi for a lot of our new pi support and you will see there is a PR here which is being discussed - https://github.com/agherzan/meta-raspberrypi/pull/783

Where there is CM4 support added. 

I will see if we have any more accurate timeline on this. At the moment I am not sure.

immiz, 2021-02-15:
> @immiz yes very much on our roadmap. We rely on the meta-raspberrypi repo here https://github.com/agherzan/meta-raspberrypi for a lot of our new pi support and you will see there is a PR here which is being discussed - [agherzan/meta-raspberrypi#783](https://github.com/agherzan/meta-raspberrypi/pull/783)
> 
> Where there is CM4 support added.
> 
> I will see if we have any more accurate timeline on this. At the moment I am not sure.

@shawaj The mentioned PR from meta-raspberrypi is merged, do you know when there will be a working balena image for the CM4?

shawaj, 2021-02-22:
@immiz i just spoke to our devices team and they are still waiting for their CM4s to arrive unfortunately. Once they have them we can begin testing and implementation on the dashboard. 

Sorry to not have a more accurate timeline, but once we have more to share I will be sure to reply on here :-)

immiz, 2021-02-22:
> @immiz i just spoke to our devices team and they are still waiting for their CM4s to arrive unfortunately. Once they have them we can begin testing and implementation on the dashboard.
> 
> Sorry to not have a more accurate timeline, but once we have more to share I will be sure to reply on here :-)

@shawaj if I can help somehow, please let me know

immiz, 2021-03-09:
ℹ️    @shawaj  I tested the CM4 Image from Balena Staging and it worked for me.
[image]

acostach, 2021-03-11:
Hi @immiz @shawaj, the image for Raspberry Pi CM4 IO Board is now available in the production dashboard.

closed 2021-03-11 by acostach

immiz, 2021-03-11:
> Hi @immiz @shawaj, the image for Raspberry Pi CM4 IO Board is now available in the production dashboard.

thx @acostach 🙏

aessig, 2021-04-09:
Hi @shawaj, thanks for the new image for the CM4 IO, testing the touch display (Official from Raspberry Pi) I realised it didn't show up, even with UDEV=1, am I missing anything here?

daniel-quantica, 2021-04-12:
I think you'll have to add the `dwc2` overlay, as the USB ports are disabled on the CM4 IO by default. At least that was what I was missing initially:

`"vc4-fkms-v3d","dwc2"`