balena-os/balena-raspberrypi #586: HDMI audio not working on Raspberry Pi 4 on 2.65.0+rev1

chrisys, 2021-01-11:
Using balenaSound to test this, HDMI audio is not working on 2.65.0+rev1 but is confirmed working on 2.50.1+rev1.

https://forums.balena.io/t/hdmi-audio-output-only-on-balenasound/232278/7

tmigone, 2021-02-12:
I think this can be closed. It is working but HDMI it's no longer "enabled by default" as an audio output since kernel version 5.4 (balenaOS v2.54.3+rev1 for Pi4's), you need to select it as the audio output interface for it to work.

I wrote a thorough explanation of this [here](https://github.com/balenalabs/balena-sound/issues/79#issuecomment-778190493), please refer to that for further details.

chrisys, 2021-02-15:
Thanks for the thorough explanation, agree that there's nothing to change here.

closed 2021-02-15 by chrisys