balena-os/balena-raspberrypi #588: Development builds can't handle enable_uart=0

nbertram, 2021-01-21:
I think this is mostly explained in the source, but the development builds have a serial console attached to the UART, so can't boot without enable_uart=1 in config.txt. This is a problem in a couple of ways.

First - in balena, via the cloud or otherwise via config vars, you can disable the UART which sets enable_uart=0. If you're not using serial, this may seem a good idea. This will cause those devices to not be able to boot (they hang at the bootloader/kernel init). If they're out of reach, that's not great, and it takes a bit of time to figure out what you did to "brick" the device.

Second - if you take a development OS device and ask it (remotely) to move to another application, it resets config.txt to a default state that omits enable_uart and reboots to begin joining the new application. Of course the device never gets past the bootloader and you're stuck. This seems unintended given the care in other places to always append enable_uart=1 to development OS config.txt to ensure it boots.

I think ultimately it would be nice if the serial console soft fails if the UART is missing so the boot process is bulletproof, but failing that a bit more guard code (probably also required in the supervisor) to stop anyone (or any process) disabling the UART on a development version of the OS could help.

Tested on BalenaOS v2.65.0+rev1.dev (raw image) with this config.txt:
`enable_uart=0
dtoverlay=vc4-fkms-v3d
arm_64bit=1
dtparam=i2c_arm=on
dtparam=spi=on
disable_splash=1
dtparam=audio=on`

(basically the out of the box config, but with uart off)

It doesn't boot at all, and the screen stays black because of disable_splash.

acostach, 2021-01-26:
Hi @nbertram, thanks for letting us know. The PR above should fix this issue.

nbertram, 2021-01-27:
OK thanks. If you'd like me to verify I'll need to build this I presume? Or is there a nightly?

acostach, 2021-01-27:
From the version number I presume you're using the Pi4, right? If so, there's a nightly build for it at https://dashboard.balena-staging.com

acostach, 2021-02-03:
This is fixed in the Pi4 release 2.67.3+rev4, feel free to reopen if the issue is still present.

closed 2021-02-03 by acostach