balena-os/balena-raspberrypi #595: Raspberry Pi CM3 support in raspberrypi3-64 image

shawaj, 2021-02-02:
Currently, the raspberrypi3-64 image does not have the .dtb for the CM3 so does not work with compute module 3 or 3+ even with the standard Compute Module IO board or other basic compute module carrier boards that do not add any features over-and-above the default.

There is a PR in the meta-raspberrypi repo to solve this https://github.com/agherzan/meta-raspberrypi/pull/783 but it is not certain how long this will take to bring in.

Perhaps we can handle this on our side until the meta-raspberrypi repo updates this? Or some of our team can help to move this along in the official repo?

See also:
[FD thread](https://www.flowdock.com/app/rulemotion/resin-devices/threads/XPP4Glb3rdlFBTtF4NZ5ROWG5Q1)
#580 

cc @floion

floion, 2021-02-05:
Fixed by https://github.com/balena-os/balena-raspberrypi/pull/594

shawaj, 2021-02-07:
Thanks @floion 👌

Also has been merged here in meta-raspberrypi https://github.com/agherzan/meta-raspberrypi/pull/783

Cc @ryanteck

ryanteck, 2021-02-22:
Hi @floion ,

Do you know when the fix will make it's way to a new OS release on Balenacloud?

Thanks

borodiliz, 2021-05-28:
@floion Any update on this? Thanks
[image]