balena-os/balena-raspberrypi #601: Community user is requesting kernel 5.10 for RPI

klutchell, 2021-02-09:
I've asked for the exact device type JF, but I would assume pi3 for now.

The new kernel has been merged upstream in master (not dunfell).

https://github.com/agherzan/meta-raspberrypi/pull/808

dtischler, 2021-02-09:
User is using a RaspberryPi 4, but, this would also be nice to have for the Fin, as he is receiving some of those as well.

krasi-georgiev, 2021-02-23:
It was me who requested it and the device is Rpi4

MTolba, 2021-03-25:
Another support [thread](https://jel.ly.fish/4c32c798-3051-445e-a3b6-46fd9cb40990) where kernel 5.10 is required to solve an issue with PCIe cards.

Resource : https://www.raspberrypi.org/forums/viewtopic.php?t=301016

dtischler, 2021-03-29:
Another email has come in (Hello@) requesting an updated kernel, albeit for the Raspberry Pi Zero...not sure if all Pi variants use same kernel or not.

closed 2021-07-22 by floion