balena-os/balena-raspberrypi #620: New BlueZ version in latest OS causes features to break.

ryanteck, 2021-03-05:
In Balena OS 2.71.3 (and possibly slightly earlier) Blue Z has been upgraded to 5.55 which introduces some breaking changes to how we were planning to use it.

It was updated here (https://github.com/balena-os/poky/commit/6fa6afd5329e9c4f80d1fc8a9365a6faacec0ed0).

Would it be possilbe to have it downgraded back to 5.53 for the NEBRA-HNT device?

ryanteck, 2021-03-07:
I can't quite tell if it's correct, but it might be as simple as adding 
```
SRC_URI_nebra-hnt[md5sum] = "e637feb2dbb7582bbbff1708367a847c"
SRC_URI_nebra-hnt[sha256sum] = "68cdab9e63e8832b130d5979dc8c96fdb087b31278f342874d992af3e56656dc"
```

to layers/meta-balena-raspberrypi/recipes-connectivity/bluez5/bluez5_%.bbappend?

floion, 2021-03-07:
Hi @ryanteck can you point to those breaking changes?

ryanteck, 2021-03-07:
Hia @floion ,

It's in relation to when using BlueZ to provide a GATT BTLE Server, in 5.55 they introduced it where it provided some data which conflicted to when user's applications provide the data.

This has now been patched at https://github.com/bluez/bluez/issues/101 which shows it is an issue. However I'm not sure in which release this patch will be deployed.

acostach, 2021-03-10:
Hi Ryan, the new image 2.71.3+rev6 is available in the dashboard and includes the patches you linked on top of the newer bluez v5.66

acostach, 2021-03-11:
Addressed by https://github.com/balena-os/balena-raspberrypi/pull/627
Feel free to re-open if necessary

closed 2021-03-11 by acostach