balena-os/balena-raspberrypi #650: Turn off all LED activity

20k-ultra, 2021-04-29:
Provide a way to disable the LEDs on the board entirely. This is not the same as telling the Supervisor to not use the LEDs. A user wants the LEDs to be completely off for all processes.

GOTO-GOSUB, 2021-04-29:
Agreed, this would be useful. A straight copy and paste from a separate issue:

> In my application I want to be able to turn off the power and activity LED's on Pi's - Pi 4 in particular. I do not mind if this takes a while to take place so long as they go off at some point and stay off. I appreciate that I may lose some debugging functionality such as LED's flashing on loss of connectivity. At the moment I have insultating tape covering the LED's which is a rather low tech solution but a simple toggle control on the dashboard would be suffice if editing the config file on the card is an issue.

20k-ultra, 2021-04-29:
@GOTO-GOSUB thanks for bringing that over, it's great context.

20k-ultra, 2021-05-01:
@GOTO-GOSUB I was able to turn off the lights on my RPi 4 after reading https://www.raspberrypi.org/forums/viewtopic.php?t=149126 and applying this to my config.txt:

```
# Disable Ethernet LEDs
dtparam=eth_led0=14
dtparam=eth_led1=14

# Disable the PWR LED
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off

# Disable the Activity LED
dtparam=act_led_trigger=none
dtparam=act_led_activelow=off
```

To be clear, to add these dtparams via the device configurations. Check out the dtparams config in https://www.balena.io/docs/learn/manage/configuration/#configuration-variables but the exact values I used were the follow (note I didn't add the ethernet LEDs though)

BALENA_HOST_CONFIG_dtparam: pwr_led_trigger=none,pwr_led_activelow=off,act_led_trigger=none,act_led_activelow=off

After doing this I tried the blink action from the dashboard and the LED does still light up in that situation but at idle the LEDs are off. I think we should close this issue now and monitor https://github.com/balena-os/balena-supervisor/issues/1375 for the final piece to this puzzle.

closed 2021-05-01 by 20k-ultra

GOTO-GOSUB, 2021-05-03:
Thank you for this, @20k-ultra . I was aware of those controls within Pi OS but had no idea how (if at all) they could be used with Balena. Excellent work, thanks for the response.