balena-os/balena-raspberrypi #665: Latest version of balenaOS Raspberry Pi for Zero stuck at 2.54.2

markFieldman, 2021-05-28:
I have some Raspberry Pi Zeros running Balena OS 2.54.2, and i want connect them to my openBalena server. 
But minimal OS version supported by openBalena is balenaOS v2.58.3.
When we will able to use Zeros with openBalena server?
When will be able to update Raspberry Pi Zeroes?

floion, 2021-05-31:
Hi, though I cannot give a definite timeline now I will say we will be working to bring it up-to-date in the next week or so.
Thanks for taking interest in it

markFieldman, 2021-05-31:
> Hi, though I cannot give a definite timeline now I will say we will be working to bring it up-to-date in the next week or so.
> Thanks for taking interest in it

Great to hear! Thanks. Alongside I trying to build OS image for my purposes, but did not achieve a good results for now. So I'm continue my tries

markFieldman, 2021-06-01:
> Hi, though I cannot give a definite timeline now I will say we will be working to bring it up-to-date in the next week or so.
> Thanks for taking interest in it

I`m success build the ing for 2.75v. But seems that image is too heavy for Zero in RAM. Will you provide some instructuions or best practics for building image for different platforms?

floion, 2021-09-20:
Released OS version `v2.83.21+rev1` for the RPI Zero

closed 2021-09-20 by floion