balena-os/balena-raspberrypi #682: User requesting v2.80.11 or newer for fincm3

klutchell, 2021-07-05:
This is the first release with supervisor v12.8.0+ as requested.

rudylacrete, 2021-07-16:
Hi there. Any update or estimate on that? This is definitely something which is slowing down our deployment and use of Balena as we need to make the update manually.

Thanks in advance for the feedback.

floion, 2021-07-16:
Hi. We are doing some updates at the moment and hope to be able to release a new version sometime next week

rudylacrete, 2021-07-28:
Hi. Do you have any update regarding next release? Thank you.

rudylacrete, 2021-08-03:
?

bnativel97400, 2021-08-16:
Hi. Do you some date about new balenaOS version with last supervisor version ?

bnativel97400, 2021-09-20:
Hey. Do you have news about next balenaOS version for fincm3 ?

floion, 2021-09-20:
Hi, sorry for the delay, we should have a new release tomorrow

floion, 2021-09-20:
Hi, we have release OS version `v2.83.21+rev1` for the Balena Fin

closed 2021-09-20 by floion