balena-os/balena-raspberrypi #695: USB GPS modules not appearing in >= 2.82.10+rev1

danthegoodman1, 2021-08-09:
Certain USB GPS modules that had worked in 2.80.5+rev1 no longer are detected by the OS in 2.82.

Downgrading is a current work around.

See forum thread:

terehov, 2021-08-16:
I literally just spend some hours trying to figure out what broke our USB devices and replaced a Raspi already. Any update on this? A downgrade feature would quite handy in such a case.

zikann, 2021-08-17:
The same issue on my device. Rpi4 8GB Serial Device and Keyboard not detected.

floion, 2021-08-18:
There is a problem with applying any overlay on 64 bits kernel `5.10.31` or newer. We'll be looking into this and will update this thread when there's progress.

floion, 2021-08-19:
Fixed by https://github.com/balena-os/balena-raspberrypi/pull/697

floion, 2021-09-03:
This is now fixed in `2.83.10+rev1` or newer

closed 2021-09-03 by floion