balena-os/balena-raspberrypi #703: USB broken on latest 2.85 version

barecool, 2021-09-02:
on 2.75 usb works fine and recognizes wifi dongles like TLW725N but on latest release usb does not work

floion, 2021-09-03:
Hi @barecool that OS release has been pulled off. Can you check the latest OS release, version `2.83.10+rev1` and let us know if the USB works on that one?

bradleywehmeier, 2021-09-09:
@barecool I was also having issues with `2.83.10+rev1`. However `2.80.5+rev1` is working fine for me.

barecool, 2021-09-12:
> Hi @barecool that OS release has been pulled off. Can you check the latest OS release, version `2.83.10+rev1` and let us know if the USB works on that one?

`2.83.10+rev1` = still doesnt work 

`2.80.5+rev1 =  works

Crowdedlight, 2021-09-13:
We also just tried to update to ``2.83.10+rev1.prod`` on our RPI4 and found the same issue. FTDI cables and our serial devices no longer show up. Downgrading to ``2.80.5+rev1.prod`` and it works again

floion, 2021-09-22:
Fixed by https://github.com/balena-os/balena-raspberrypi/pull/712