balena-os/balena-raspberrypi #715: Raspberry Pi PoE+ hat fan not workingAgreed with @monsterslo; using a Raspberry 4 with Raspberry's own POE+ board.  Using supervisor 12.10.1 and balenaOS 2.83.10+rev1.

HardwareHarry, 2021-10-01:
Creating a new issue as I realized the other open issue was for the original Pi PoE hat, not the newer PoE+ hat and actually both recent comments had been made for PoE+ not PoE (so issue #339 can probably be closed now).

Using supervisor 12.10.1 and balenaOS 2.83.10+rev1, the PoE+ hat doesn't ever seem to spin, even when the CPU temperature is 70+ degrees centigrade.  The overlay that covers the poe+ appears to be missing (though I can see the one for the original hat - rpi-poe.dtbo).  I believe the new hat's overlay is rpi-poe-plus.dtbo.

_Originally posted by @HardwareHarry in https://github.com/balena-os/balena-raspberrypi/issues/339#issuecomment-917422895_

floion, 2021-12-14:
Should be fixed by https://github.com/balena-os/balena-raspberrypi/pull/744

luke-falvey, 2022-01-13:
For those that need a temporary solution while this isn't released

1. Download rpi-poe-plus firmware at https://github.com/raspberrypi/firmware/blob/master/boot/overlays/rpi-poe-plus.dtbo

2. Place it under /boot/overlays on the SD card

3. Add "rpi-poe-plus" to the "Define DT parameters" in Device Configuration in balena-cloud. My configuration looks like "vc4-fkms-v3d","rpi-poe-plus"

4. Reboot raspberry pi 4

My fans started spinning after a few minutes

HardwareHarry, 2022-01-14:
@luke-falvey do you see the following output from dmesg?
rpi-poe-power-supply rpi-poe-power-supply@0: Unsupported firmware
rpi-poe-power-supply: probe of rpi-poe-power-supply@0 failed with error -2

I tried loading your steps - I can't check the actual fan as it's mounted inside a weatherproof box on my roof, but I do see the module being loaded.  It isn't populating /sys/devices/platform/rpi-poe-power-supply@0 - which is a shame as I have both a POE+ board and a Pi Juice board both attached to the Pi, and the PiJuice is constantly complaining that +5v is either "bad", "low" or "failed" (it seems to be hovering about the 4.8-4.9v) - I'm trying to work out whether the POE+ is really providing low VCC or the PiJuice's monitoring is askew.

floion, 2022-03-10:
@luke-falvey @HardwareHarry this should be fixed now in the latest OS release for the PI4. Can you confirm and close this ticket if fixed? Thanks

HardwareHarry, 2022-03-10:
Thanks @floion, I can confirm that I upgraded to the latest release of both supervisor and OS and the fan does now function correctly, also the rpi-poe-plus.dbto is present and loaded and the various /sys entries show correct information.  Closing the issue, thanks so much for fixing!

closed 2022-03-10 by HardwareHarry