balena-os/balena-raspberrypi #731: Bluetooth Controller - MAC Address Appears as AA:AA:AA:AA:AA:AA

dtischler, 2021-11-17:
A user is reporting that on a CM4, Bluetooth is generating a MAC Address of AA:AA:AA:AA:AA:AA (and sometimes does not come up at all).

User is on balenaOS 2.85.2+rev3.  User reports that on Raspbian, it works properly.

``
root@artsy-sea:~# bluetoothctl
Agent registered
[CHG] Controller AA:AA:AA:AA:AA:AA Pairable: yes
AdvertisementMonitor path registered
[bluetooth]#
```

@markcorbinuk tested and got similar results on his Pi, and AlexG asked for an Issue.  Mark also linked to these threads:

https://forums.raspberrypi.com/viewtopic.php?t=304000&start=25
https://forums.raspberrypi.com/viewtopic.php?t=272448

mhazley, 2021-11-24:
Hi, I'm the user that reported the issue. I've been digging into it today, posting some notes below in case they are useful. 

With regards to the device coming up all the time, if I set `dtparam=krnbt=on`- Bluetooth comes up consistently and without fail. 

With regards to the mac address, I can see that I can set it manually using hcitool:

```
# hcitool cmd 0x3f 0x001 0x01 0x02 0x03 0x06 0x05 0x06
< HCI Command: ogf 0x3f, ocf 0x0001, plen 6
  01 02 03 06 05 06
> HCI Event: 0x0e plen 4
  01 01 FC 00
```

This led me to dig into how RPi sets it when using `krnbt=on`.

They use a script `/usr/bin/bthelper` but it doesn't seem to work for me when I run it manually, it just claims that the address it already set. 

```
/usr/bin/bthelper hci0
Raspberry Pi BDADDR already set
```

Maybe there's something here?

acostach, 2021-11-27:
Indeed @mhazley , looks like the switch back to `'\s43:4[35]:' `in upstream pi-bluetooth causes this, the old check for `\s(B8:27:EB:|DC:A6:32:)` works fine on the cm4-ioboard. I've added a patch in the PR above for this.

acostach, 2021-11-29:
@mhazley I'll wait a bit before merging this, so the upstream maintainer of pi-bluetooth can confirm if this is the right approach

acostach, 2021-12-09:
Hi @mhazley , a new image v2.88.4 is now available for the CM4 in the dashboard. We haven't observed any issue with BT coming up, but if you need to set krnbt you can do that from the dashboard, in the device configuration page.

closed 2021-12-09 by acostach