balena-os/balena-raspberrypi #736: Request for Quectel BG95-M3 modem on Raspberry Pi 3

klutchell, 2021-11-25:


Mk-arc, 2021-11-26:
Hi,

I am working at Quectel and currently supporting the customer who raised this issue. Let me explain my understanding of this Problem:

Problem is the in-tree "**option driver**", handling serial usb devices. The BG95-M3 has the PID and VID: "2c7c 0700".

It seems now that the standard option driver in your balea os is not correctly assigning the interface to this USB device. Causing the following issue.

The BG95 registers correctly at the USB bus... its appearing in "lsusb". But the option driver will not grab interfaces, meaning no devices appear in /dev/ttyUSB*. The BG95 should create 4 serial ttyUSB devices.

When can manually assign the device identifier to the option driver:

#!/bin/bash
sudo modprobe option
echo "2c7c 0700" | sudo tee -a /sys/bus/usb-serial/drivers/option1/new_id

This will fix the issue. Devices under /dev/ttyUSB* are then loaded correctly.

So what you need is not an additional driver, but a patched option driver that supports this modem.

Quectel provides this patches already:
https://cnquectel-my.sharepoint.com/:u:/g/personal/europe-fae_quectel_com/EfosTDMCu7dKtvLl5_XsYf4BDBIawUjtoRen6cLPlJdJ-Q?e=18V5GI

You just need to apply them. The driver basically just adds all the identifier of quectel products to the driver:

in option.c
```
static const struct usb_device_id option_ids[] = {
#if 1 //Added by Quectel
	{ USB_DEVICE(0x05C6, 0x9090) }, /* Quectel UC15 */
	{ USB_DEVICE(0x05C6, 0x9003) }, /* Quectel UC20 */
	{ USB_DEVICE(0x05C6, 0x9215) }, /* Quectel EC20(MDM9215) */
	{ USB_DEVICE(0x2C7C, 0x0125) }, /* Quectel EC20(MDM9x07)/EC25/EG25 */
	{ USB_DEVICE(0x2C7C, 0x0121) }, /* Quectel EC21 */
	{ USB_DEVICE(0x2C7C, 0x0191) }, /* Quectel EG91 */
	{ USB_DEVICE(0x2C7C, 0x0195) }, /* Quectel EG95 */
	{ USB_DEVICE(0x2C7C, 0x0306) }, /* Quectel EG06/EP06/EM06 */
	{ USB_DEVICE(0x2C7C, 0x0512) }, /* Quectel EG12/EP12/EM12/EG16/EG18 */
	{ USB_DEVICE(0x2C7C, 0x0296) }, /* Quectel BG96 */
	{ USB_DEVICE(0x2C7C, 0x0700) }, /* Quectel BG95/BG77/BG600L-M3/BC69 */
	{ USB_DEVICE(0x2C7C, 0x0435) }, /* Quectel AG35 */
	{ USB_DEVICE(0x2C7C, 0x0415) }, /* Quectel AG15 */
	{ USB_DEVICE(0x2C7C, 0x0520) }, /* Quectel AG520 */
	{ USB_DEVICE(0x2C7C, 0x0550) }, /* Quectel AG550 */
	{ USB_DEVICE(0x2C7C, 0x0620) }, /* Quectel EG20 */
	{ USB_DEVICE(0x2C7C, 0x0800) }, /* Quectel RG500/RM500/RG510/RM510 */
	{ USB_DEVICE(0x2C7C, 0x6026) }, /* Quectel EC200 */
	{ USB_DEVICE(0x2C7C, 0x6120) }, /* Quectel UC200 */
	{ USB_DEVICE(0x2C7C, 0x6000) }, /* Quectel EC200/UC200 */
#endif
```

also in option.c handling the seperate interfaces of the quectel devices:
```
#if 1 //Added by Quectel
	.reset_resume = usb_wwan_resume,
#endif
#endif
};

static struct usb_serial_driver * const serial_drivers[] = {
	&option_1port_device, NULL
};

module_usb_serial_driver(serial_drivers, option_ids);

static int option_probe(struct usb_serial *serial,
			const struct usb_device_id *id)
{
	struct usb_interface_descriptor *iface_desc =
				&serial->interface->cur_altsetting->desc;
	unsigned long device_flags = id->driver_info;

#if 1 //Added by Quectel
	//Quectel UC20's interface 4 can be used as USB Network device
	if (serial->dev->descriptor.idVendor == cpu_to_le16(0x05C6) && serial->dev->descriptor.idProduct == cpu_to_le16(0x9003)
		&& serial->interface->cur_altsetting->desc.bInterfaceNumber >= 4)
		return -ENODEV;

	//Quectel EC20(MDM9215)'s interface 4 can be used as USB Network device
	if (serial->dev->descriptor.idVendor == cpu_to_le16(0x05C6) && serial->dev->descriptor.idProduct == cpu_to_le16(0x9215)
		&& serial->interface->cur_altsetting->desc.bInterfaceNumber >= 4)
		return -ENODEV;

	if (serial->dev->descriptor.idVendor == cpu_to_le16(0x2C7C)) {
		__u16 idProduct = le16_to_cpu(serial->dev->descriptor.idProduct);

		//Quectel module's some interfaces can be used as USB Network device (ecm, rndis, mbim)
		if (serial->interface->cur_altsetting->desc.bInterfaceClass != 0xFF)
			return -ENODEV;

		//Quectel EC25&EC20's interface 4 can be used as USB network device (qmi)
		if ((idProduct != 0x6026 && idProduct != 0x6120) && serial->interface->cur_altsetting->desc.bInterfaceNumber >= 4)
			return -ENODEV;
	}
#endif
```

This is my understanding why the BG95 does not work without manually assigning the device to the option driver in BaleaOS.

Mk-arc, 2021-11-26:
I forgot, here is also a nice User Guide that describes the changes needed:

[Quectel_LTE&5G_Linux_USB_Driver_User_Guide_V2.0.pdf](https://github.com/balena-os/balena-raspberrypi/files/7606709/Quectel_LTE.5G_Linux_USB_Driver_User_Guide_V2.0.pdf)

floion, 2021-12-07:
Hi, thanks for the details @Mk-arc . Will have a look in the following days

floion, 2021-12-14:
Hi @Mk-arc quick follow-up question. What would be Quectel's recommendation on what driver to use? My personal option would be to have it use the qmi driver but I see you detailed your answer on how to add the changes for the option driver. Wouldn't it be better, from a performance standpoint, to have the modem working in qmi mode?

Mk-arc, 2021-12-14:
Hi @floion 

It needs to use multiple drivers. Like with most Quectel Modules. 

The **option** driver is needed to create the ports for at command communication, debugging, nmea interface and for some modules ADB port and modem port. The option driver needs to grab these interfaces and then open four (or 3) ttyUSB devices. Other interfaces need to be grabbed by other drivers. Thats why in the option driver it will only catch the first few ports. So that others are free to use for other drivers, this change is important.

For BG95: **This module has no QMI support**. (most other Qualcomm based Quectel Modules do have it). For BG95 the **cdc_ether driver (ECM)** is used for data communication (however, it is disabled by default) or PPP.

floion, 2022-01-13:
This will be available next in release v2.88.5+rev1

acostach, 2022-03-29:
Closing, newer images v2.9x images are now available

closed 2022-03-29 by acostach