balena-os/balena-raspberrypi #746: bitbake fails due to git segfault on Rpi-distro

duststorm, 2021-12-15:
Maybe not entirely a bug for this repository, but at least useful to document this failing configuration.
When building on Ubuntu 20.04
Git is at version 1:2.25.1-1ubuntu3.2

```
git --version
git version 2.25.1
```
bitbake balena-image fails with the following error:
```
    WARNING: linux-firmware-rpidistro-20190114-1+rpt10-r0 do_fetch: Failed to fetch URL git://github.com/RPi-Distro/firmware-nonfree, attempting MIRRORS if available
    ERROR: linux-firmware-rpidistro-20190114-1+rpt10-r0 do_fetch: Fetcher failure: Unable to find revision b66ab26cebff689d0d3257f56912b9bb03c20567 in branch master even from upstream
    ERROR: linux-firmware-rpidistro-20190114-1+rpt10-r0 do_fetch: Fetcher failure for URL: 'git://github.com/RPi-Distro/firmware-nonfree'. Unable to fetch URL from any source.
    ERROR: Logfile of failure stored in: /home/duststorm/balena-raspberrypi/build/tmp/work/all-poky-linux/linux-firmware-rpidistro/20190114-1+rpt10-r0/temp/log.do_fetch.1127018
    ERROR: Task (/home/duststorm/balena-raspberrypi/build/../layers/meta-raspberrypi/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb:do_fetch) failed with exit code '1'
    NOTE: Tasks Summary: Attempted 2512 tasks of which 0 didn't need to be rerun and 1 failed.
```

The following happens when the recipes
layers/meta-raspberrypi/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb
layers/meta-balena-raspberrypi/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_%.bbappend
are being executed:

```
git -c core.fsyncobjectfiles=0 clone --bare --mirror "git://github.com/RPi-Distro/firmware-nonfree" --progress
git -c core.fsyncobjectfiles=0 branch --contains b66ab26cebff689d0d3257f56912b9bb03c20567 --list master 2> /dev/null | wc -l
# which returns 0
    
git -c core.fsyncobjectfiles=0 remote
# which returns origin
git -c core.fsyncobjectfiles=0 remote rm origin
git -c core.fsyncobjectfiles=0 remote add --mirror=fetch origin "git://github.com/RPi-Distro/firmware-nonfree"
git -c core.fsyncobjectfiles=0 fetch -f --progress "git://github.com/RPi-Distro/firmware-nonfree" refs/*:refs/*
git -c core.fsyncobjectfiles=0 prune-packed
git -c core.fsyncobjectfiles=0 pack-refs --all
git -c core.fsyncobjectfiles=0 pack-redundant --all | xargs -r rm
# Here things go wrong:
# The following command leads to a segfault
git -c core.fsyncobjectfiles=0 pack-redundant --all
Segmentation fault (core dumped)

# Following this, the next steps fail
git -c core.fsyncobjectfiles=0 branch --contains b66ab26cebff689d0d3257f56912b9bb03c20567 --list master 2> /dev/null | wc -l
# again returns 0
# which is why bitbake spits out:

WARNING: Failed to fetch URL git://github.com/RPi-Distro/firmware-nonfree, attempting MIRRORS if available
DEBUG: Fetcher failure: Unable to find revision b66ab26cebff689d0d3257f56912b9bb03c20567 in branch master even from upstream
NOTE: Removing /home/duststorm/balena-raspberrypi/build/downloads/git2/github.com.RPi-Distro.firmware-nonfree
DEBUG: Trying MIRRORS

# and goes on to download a tar.gz from openembedded.org, which again fails because of the same git crash:
DEBUG: Mirror fetch failure for url http://sources.openembedded.org/git2_github.com.RPi-Distro.firmware-nonfree.tar.gz (original url: git://github.com/RPi-Distro/firmware-nonfree)
DEBUG: Fetcher failure: Unable to find revision b66ab26cebff689d0d3257f56912b9bb03c20567 in branch master even from upstream
```

Anyone a clue what git problem this could be related to and whether it is already fixed upstream?
What reference environment is used to perform the official balena builds?

duststorm, 2021-12-15:
The plot thickens.
If I update git to version 2.34

I get the following message on the command that previously caused the crash:
```
git -c core.fsyncobjectfiles=0 pack-redundant --all
'git pack-redundant' is nominated for removal.
If you still use this command, please add an extra
option, '--i-still-use-this', on the command line
and let us know you still use it by sending an e-mail
to <git@vger.kernel.org>.  Thanks.
```

If I run the command as follows I no longer get a crash:
```
git -c core.fsyncobjectfiles=0 pack-redundant --i-still-use-this --all
```
However, it appears to change nothing, the command after that still yields no results:
```
git -c core.fsyncobjectfiles=0 branch --contains b66ab26cebff689d0d3257f56912b9bb03c20567 --list master 2> /dev/null | wc -l
# returns 0
```
So maybe the segfault is not the cause of the issue

The recipe that executes this command seems to stem from poky:
```
layers/poky/bitbake/lib/bb/fetch2/git.py:            runfetchcmd("%s pack-redundant --all | xargs -r rm" % ud.basecmd, d, workdir=ud.clonedir)
```

Maybe my issue is similar to this one:
https://github.com/riscvarchive/riscv-poky/issues/15

duststorm, 2021-12-21:
The build works fine with tag v2.88.4 and master branch
I was confused by the fact that the releases page of this repo appears to be outdated.

However, building 2.25.1 is broken due to the git checkout issue in meta-raspberrypi (yocto).
Perhaps I should report this issue over there.

floion, 2022-01-18:
Hi,
sorry for the late reply.
The git checkout issue for that old release should be reported upstream. But I think there's not much they can backport anyway if the upper git repo has changed history (as it seems to me).
Any particular reason you need to build such an old OS version? (2.25.1)

floion, 2022-03-10:
@duststorm ?