balena-os/balena-raspberrypi #749: [RPI4] ffmpeg fails to play rtsp stream using h264_v4l2m2m

acostach, 2021-12-17:
This behavior is caused by changes in upstream rpi-linux bcm2835-codec.

Steps to reproduce:

- Provision device on balenaOS v2.88.4 or any other release that uses RPI kernel 5.10.78
- Configure gpu_mem to 256 or higher so that /dev/video10,11,12... get created
- Run the following container from the hostOS webterminal:
```
balena run --rm \
  --device /dev/video10 \
  --device /dev/video11 \
  --device /dev/video12 \
  --device /dev/bus/usb \
  --entrypoint ffmpeg linuxserver/ffmpeg \
  -c:v h264_v4l2m2m -i "rtsp://rtsp.stream/pattern" -filter:v fps=1 -f rawvideo -pix_fmt nv12 test.out
```

The above will fail to decode with:
`Error while decoding stream #0:0: Resource temporarily unavailable`

Kernel 5.10.31 does not exhibit this issue, and there are 18 commits in the bcm2835-codec kernel module between 5.10.31 and 5.10.78. Reverting all these commits makes the problem go away:

```
[h264_v4l2m2m @ 0x557de2ae30] Using device /dev/video10
[h264_v4l2m2m @ 0x557de2ae30] driver 'bcm2835-codec' on card 'bcm2835-codec-decode' in mplane mode
[h264_v4l2m2m @ 0x557de2ae30] requesting formats: output=H264 capture=YU12
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_v4l2m2m) -> rawvideo (native))
Press [q] to stop, [?] for help
Output #0, rawvideo, to 'test.bin':
  Metadata:
    title           : Stream
    encoder         : Lavf58.76.100
  Stream #0:0: Video: rawvideo (NV12 / 0x3231564E), nv12(tv, progressive), 320x240, q=2-31, 921 kb/s, 1 fps, 1 tbn
    Metadata:
      encoder         : Lavc58.134.100 rawvideo
^C^C^C^CReceived > 3 system signals, hard exiting0:00:03.00 bitrate= 699.1kbits/s speed=1.16x    
root@4d0a24f:~# uname -r
5.10.78-v8
```

List of reverted changes that makes this work again below. Need to tested to find out which one(s) is the exact root cause and if this is really a problem in the driver or in ffmpeg:

```
file://0002-Revert-bcm2835-v4l2-codec-Remove-advertised-support-.patch \
file://0003-Revert-media-bcm2835-codec-Limit-video-callbacks.patch \
file://0004-Revert-staging-bcm2835-codec-Add-support-for-H-V-Fli.patch \
file://0005-Revert-staging-bcm2835-codec-Ensure-all-ctrls-are-se.patch \
file://0006-Revert-staging-bcm2835_codec-Correct-flushing-code-f.patch \
file://0007-Revert-staging-bcm2835-codec-Queue-flushed-buffers-i.patch \
file://0008-Revert-staging-bcm2835-codec-Signal-the-firmware-to-.patch \
file://0009-Revert-staging-bcm2835-codec-Format-changed-should-t.patch \
file://0010-Revert-staging-bcm2835-codec-Do-not-send-buffers-to-.patch \
file://0011-Revert-staging-bcm2835-codec-Allow-decode-res-change.patch \
file://0012-Revert-staging-bcm2835-v4l2_codec-Fix-for-encode-sel.patch \
file://0013-Revert-staging-bcm2835_codec-Add-support-for-image_f.patch \
file://0014-Revert-staging-bcm2835-codec-Allow-custom-specified-.patch \
file://0015-Revert-staging-bcm2835_codec-Log-MMAL-flags-in-hex.patch \
file://0016-Revert-staging-bcm2835-codec-Return-buffers-to-QUEUE.patch \
file://0017-Revert-staging-bcm2835-codec-Correct-ENUM_FRAMESIZES.patch \
file://0018-Revert-staging-bcm2835-codec-Add-support-for-decodin.patch \
file://0019-Revert-staging-bcm2835-codec-Change-the-default-code.patch \
```

zwhitchcox, 2021-12-17:
Seems like this might be fixed by https://github.com/raspberrypi/linux/pull/4773

zwhitchcox, 2021-12-17:
See https://github.com/raspberrypi/linux/pull/4658#issuecomment-954794183

acostach, 2021-12-19:
With this change https://github.com/raspberrypi/linux/commit/b7e6b495eff31298ba4665f71b2414cc9a8f99c2 reverted,  the problem is no longer present. Tested on 5.10.78 and 5.10.87 (latest)

acostach, 2021-12-21:
Waiting for feedback from upstream maintainer in https://github.com/raspberrypi/linux/issues/4785

acostach, 2022-02-03:
https://jel.ly.fish/869a4c1b-866c-4509-b508-635eee2988c2 should be fixed by updating ffmpeg in container as per upstream ticket: https://github.com/raspberrypi/linux/issues/4785 (which is not the root cause) to include ffmpeg fix https://github.com/FFmpeg/FFmpeg/commit/3fc72c9fc16df3b7edc329ed9eeaaa1e65976c33

https://jel.ly.fish/b6a19a2f-4297-4e10-9b55-34b14273ebe2 doesn't look related to this issue and might have been attached by mistake.

closed 2022-02-03 by acostach