balena-os/balena-raspberrypi #753: User request for v2.88 or newer for Raspberry Pi Zero 2

maggie0002, 2022-01-01:
v2.88 incorporated import NTP time sync checks that I need to be able to get the Zero 2 working efficiently. At the moment I get SSL timeouts when using online features after immediate setup of a device due to the dates being wrong. 

Logging an issue for this request as suggested here: https://forums.balena.io/t/notifications-when-balenaos-updates-are-released/350287/10

maggie0002, 2022-01-25:
Also looking for 2.88.13 or up for the fix to the development images incorrectly changing to production: https://forums.balena.io/t/balena-raspberry-pi-os-dev-is-actually-a-prod-versions/349857/35

acostach, 2022-03-07:
Hi @maggie0002 , v2.94.4 has been released for Pi Zero 2W.

closed 2022-03-07 by acostach