balena-os/balena-raspberrypi #764: Add support for NVME boot for CM4 based boards

anujdeshpande, 2022-01-17:


rosswesleyporter, 2022-01-18:
balenaOS on a CM4 IO Board cannot boot from NVMe. But Pi OS can boot from NVMe. I'm wondering if balenaOS has a sufficiently recent version of the VideoCore firmware.

[According to the Pi Foundation instructions](https://github.com/raspberrypi/documentation/blob/develop/documentation/asciidoc/computers/raspberry-pi/boot-nvme.adoc), booting from NVMe requires an updated bootloader and updated VideoCore firmware. I updated the bootloader in EEPROM and Pi OS booted from NVMe. I then tried balenaOS (v2.88.4+rev0), but it failed to boot, generating this UART output:

```
Hit any key to stop autoboot: 0 
switch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
Card did nt respond to voltage select! : -110
Scanning disk mmcnr@7e300000.blk...
Disk mmcnr@7e300000.blk not ready
Scanning dsk mmc@7e340000.blk...
Found 3 disks
No EFI system partition
BootOrder not defined
EFI boot manager: Cannot load any image
```

I used the current (October 30th 2021) version of Pi OS, which presumably has a sufficiently recent version of the VideoCore firmware on the boot media. I don't know what version of the VideoCore firmware is used by balenaOS.

floion, 2022-03-10:
Should be fixed in `2.94.4+rev1`
@rosswesleyporter can you test this and close this ticket if fixed?
Thanks

rosswesleyporter, 2022-03-10:
Tested balenaOS 2.94.4+rev1 on staging. Was able to boot from a Samsung SSD 980 NVMe drive as well as from a Kingspec NVMe drive. But note that I manually updated the bootloader in the CM4 EEPROM. I believe that manually updating EEPROM will be necessary until the CM4 ships with an updated EEPROM.