balena-os/balena-raspberrypi #766: Raspberry Pi 4 download link broken

wyuenho, 2022-01-17:
This link on the [balenaOS](https://api.balena-cloud.com/download?deviceType=raspberrypi4-64&version=2.88.5+rev1.prod&fileType=.zip&d_id=57bb8927-2111-47d0-b8e4-8bf1fa298946R) download page is currently returning 500.

floion, 2022-01-18:
Hi, how did you get that link?
Starting with this version, .prod and .dev images have been unified. So your link would be something like this now: https://api.balena-cloud.com/download?deviceType=raspberrypi4-64&version=2.88.5+rev1&fileType=.zip&d_id=57bb8927-2111-47d0-b8e4-8bf1fa298946R

wyuenho, 2022-01-18:
It's just the link listed at https://www.balena.io/os/

airdrummingfool, 2022-01-22:
@floion do you have any documentation about that change? I can't find anything, and it's confusing to download a dev image that isn't marked as such (but seems to still behave as such).

floion, 2022-03-09:
@wyuenho @airdrummingfool it's been fixed now as you probably saw.

floion, 2022-03-09:
Can this be closed now?

closed 2022-03-10 by wyuenho