balena-os/balena-raspberrypi #779: Bluetooth interoperability problems with some newer mobile phones

alexgg, 2022-02-05:
Customers implementing a GATT server application have reported communication problems with clients running on newer mobile phones like the Samsung S20.

The problems have been traced to a firmware revision change:

On v2021.10.2, the following exhibit the problem:
```
linux-firmware-rpidistro

cat balena-image-raspberrypi4-64-2021.10.2.dev.manifest | grep linux-firmware-rpidistro
linux-firmware-rpidistro-bcm43430 all 20190114-1+rpt10-r0
linux-firmware-rpidistro-bcm43455 all 20190114-1+rpt10-r0
linux-firmware-rpidistro-broadcom-license all 20190114-1+rpt10-r0

bluez-firmware-rpidistro

cat balena-image-raspberrypi4-64-2021.10.2.dev.manifest | grep bluez-firmware-rpidistro
bluez-firmware-rpidistro-bcm43430a1-hcd all 1.2+git0+e7fd166981-r0
bluez-firmware-rpidistro-bcm4345c0-hcd all 1.2+git0+e7fd166981-r0
bluez-firmware-rpidistro-cypress-license all 1.2+git0+e7fd166981-r0
```
While on v2020.07.2, the following don't:
```
linux-firmware-rpidistro

cat resin-image-raspberrypi4-64-2020.07.2.dev.manifest | grep linux-firmware-rpidistro
linux-firmware-rpidistro-bcm43430 all 0.0+git0+f0ad1a42b0-r0
linux-firmware-rpidistro-bcm43455 all 0.0+git0+f0ad1a42b0-r0
linux-firmware-rpidistro-broadcom-license all 0.0+git0+f0ad1a42b0-r0

bluez-firmware-rpidistro

cat resin-image-raspberrypi4-64-2020.07.2.dev.manifest | grep bluez-firmware-rpidistro
bluez-firmware-rpidistro-bcm43430a1-hcd all 0.0+git0+96eefffccc-r0
bluez-firmware-rpidistro-bcm4345c0-hcd all 0.0+git0+96eefffccc-r0
bluez-firmware-rpidistro-cypress-license all 0.0+git0+96eefffccc-r0
```

alexgg, 2022-02-05:
Could be related to https://github.com/RPi-Distro/bluez-firmware/issues/9

alexgg, 2022-02-05:
Changelog between both versions:
```
e7fd166981ab (HEAD -> master, origin/master, origin/HEAD) 1.2-4+rpt8 release
8445a53ce2c5 Second Spectra fix for CYW43455 (CVE-2020-10370)
1e4ee0c05bae 1.2-4+rpt7 release
da4bfb1ecf41 Bug fix CYW43455 firmware
a4e08822e3f2 1.2-4+rpt6 release
929b7af9c43e Add BCM43430B0.hcd
c317a5c56847 1.2-4+rpt5 release
c3cab65cec7d 1.2-4+rpt4 release
e75a3f546a83 1.2-4+rpt3 release
ba01b70ddf3f Update to BCM43455 firmware with Spectra fixes
afe608e7055a Update to BCM43438 firmware with Spectra fixes
2e64c6624aa1 Add BCM4345C5.hcd
98cbd4489727 Post-Sweyntooth update of BCM43430A1 and BCM4345C0 hcds
fff76cb15527 (origin/buster) 1.2-4+rpt2 release
83921f41fe58 Rebase on Buster release
```

alexgg, 2022-02-05:
https://community.infineon.com/t5/Security-Bulletin/Security-Bulletin-Potential-Wi-Fi-Bluetooth-Combo-Security/ba-p/246960

mhazley, 2022-04-26:
Hey @alexgg I have encountered this problem on RPi3 using a Pixel 6 - I wondered if you could advise if this change is in latest RPi OS's or just the 2021.10.4 ESR, as I don't see that ESR for RPi 3.