balena-os/balena-raspberrypi #803: No analog audio support in 2.91

justinasjaronis, 2022-03-01:
There has been a breaking change in 2.91 that prevents using analog audio out (headphone jack).

In earlier versions (i.e. 2.88.5) default configuration provides /dev/snd/by-path/platform-bcm2835_audio

alsa show one soundcard with many outputs:
bash-5.1# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7

However in newest version multiple cards for both hdmi ports are visible (I believe this is change from upstream kernel), but there is no analog audio card by default.

ls /dev/snd/by-path/platform-fef0
platform-fef00700.hdmi  platform-fef05700.hdmi  

Furthermore kernel module snd_bcm2835 is not loaded at all by default. I tried to load with various module parameters but with no luck. BTW pulseaudio 

What's possible to do to get analog sound in newer images, if it's not on in default?

justinasjaronis, 2022-03-01:
So the problem with newest release is that if we are using vc4-kms-v3d overlay, we are having just two hdmi sound devices in /dev/snd/by-path/. If we do not load this overlay or load vc4-fkms-v3d , we have /dev/snd/by-path/platform-bcm2835_audio . I think this is breaking change that wasn't present in earlier versions.

How do we use sound  via headphone jack when loading vc4-kms-v3d overlay?

justinasjaronis, 2022-03-01:
I found this interesning information:

 <...> I figured something out that I have not seen pointed out anywhere that I recall. But if you use dtparam=audio=on in your config.txt, it does different things based on its relative position to dtoverlay=vc4-kms-v3d-pi4.

If it follows the dtoverlay=vc4-kms-v3d-pi4 and before another dtoverlay command, it becomes a parameter to the dtoverlay, and acts to turn on digital audio. Even if you set dtoverlay=vc4-kms-v3d-pi4,noaudio=on (which turns digital audio off).

However, if dtparam=audio=on is placed before dtoverlay=vc4-kms-v3d-pi4 it turns on analog audio via bcm2835.

So I turn on analog audio and turn off digital audio:

dtparam=audio=on
dtoverlay=vc4-kms-v3d-pi4,noaudio=on

I can confirm that those parameters in opposite order doesn't fix analog audio issue.

acostach, 2022-03-02:
Hi, what device is this for? Is it an image from the dashboard or one built by yourself?

config.txt elemens are appended by https://github.com/balena-os/balena-raspberrypi/blob/master/layers/meta-balena-raspberrypi/recipes-bsp/bootfiles/rpi-config_git.bbappend on top of the upstream recipe https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-bsp/bootfiles/rpi-config_git.bb

justinasjaronis, 2022-03-02:
I'm talking about vanilla images from the balena-cloud dashboard.  Raspberry Pi4 64bit

acostach, 2022-03-02:
I see, thanks for clarifying. In v2.88.5 vc4-kms-v3d was not getting loaded, because in upstream the overlay-map was modified to load vc4-kms-v3d-pi4.dtbo instead of vc4-kms-v3d.dtbo and the v2.88.5 image did not include this newer overlay. So the overlay was not getting loaded at all and analog audio worked. The v2.91.1 image uses the upstream raspberrypi linux version 5.10.83 and includes both vc4-kms-v3d-pi4.dtbo and vc4-fkms-v3d-pi4.

I suggest trying to reproduce the issue with the latest RaspberryPi OS. If it isn't reproducible and the kernel is newer than 5.10.83 we update to that version, otherwise it needs to be reported in upstream.