balena-os/balena-raspberrypi #895: raspberrypi3-64: Subtest: Engine socket is not exposed in production images

klutchell, 2022-07-12:
```
# Subtest: Engine socket is not exposed in production images
            # Setting system in production mode...
/data/suite/node_modules/ext2fs/lib/libext2fs.js:1
```
which means we have some unhandled error. The failure here seems exclusive to this device type right now

rcooke-warwick, 2022-07-13:
Looking at the stack trace, past the big dump of libext2fs logs, we might be having an issue with dockerode:

```
Error: 
    at /data/suite/tests/engine-socket/index.js:49:15
    at /data/suite/node_modules/dockerode/lib/docker.js:1222:7
    at Modem.buildPayload (/data/suite/node_modules/docker-modem/lib/modem.js:253:19)
    at ClientRequest.<anonymous> (/data/suite/node_modules/docker-modem/lib/modem.js:238:10)
    at ClientRequest.emit (events.js:400:28)
    at Socket.socketOnEnd (_http_client.js:499:9)
    at Socket.emit (events.js:412:35)
    at endReadableNT (internal/streams/readable.js:1333:12)
    at processTicksAndRejections (internal/process/task_queues.js:82:21) {
  tapCaught: 'uncaughtException',
  test: 'Engine socket exposure test'
  ```

Dockerode is used exclusively in these engine socket tests. Still trying to figure out why its only happening on the rpi3-64 tests.

rcooke-warwick, 2022-07-13:
identified that this is indeed the problem ^ working on a fix

closed 2022-07-23 by bulldozer-balena[bot]