balena-os/balena-raspberrypi #924: revpi-connect-can: missing device after upgrade to balenaOS 2.98.33

adolfogc, 2022-10-15:
Hi,

I recently upgraded a device to balenaOS 2.98.33 and now it get's stuck in a watchdog-induced reboot-loop due to a missing revpi-specific device:

```bash
No such file or directory: '/dev/piControl0'
```

I noticed that the `/mnt/boot/config.txt' file was missing after the upgrade (but it does exist in the image, which I downloaded and inspected).

Any ideas on what may had gone wrong?

wickywaka, 2023-01-10:
Did you manage to solve this?

floion, 2023-04-29:
On what device type did this issue appear? What hardware and what OS version and type?

adolfogc, 2023-05-03:
Hi, @floion  the device is currently running balenaOS (production) version 2.80.3+rev1, on a Revpi Connect. The conflicting version is the one stated in the issue's title (2.98.33) --I didn't try later versions. @wickywaka You can use version 2.80.3+rev1 for now.