balena-os/balena-raspberrypi #945: gpio-poweroff.dtbo sometimes missing from dtbo list

acostach, 2022-11-30:
Some images pack gpio-poweroff.dtbo, sometimes don't.This seems to be caused by a race at the time the overlays are deployed and their list is created. Investigated in https://github.com/balena-os/balena-raspberrypi/pull/943

Bucknalla, 2022-12-13:
Hey @acostach #943 seems to be merged; did that fix this issue?

acostach, 2022-12-13:
Hi @Bucknalla , yes, the issue was fixed with that PR. We added a check at build time to abort if the overlay isn't present in the image and I haven't seen any build failures due to that.

There haven't been automatic deploys for a while though, so the fix is not included in any new image release for the Pi4 - see internal thread https://balena.zulipchat.com/#narrow/stream/345889-loop.2Fbalena-os/topic/.E2.9C.94.20No.20OS.20releases/near/315572933

acostach, 2023-01-14:
This has been fixed in production starting with v2.108.18

closed 2023-01-14 by acostach