balena-os/balena-raspberrypi #999: Missing imx708 overlay

Jessseee, 2023-03-08:
I am having trouble getting the new Raspberry Pi Camera Module v3 to work on BalenaOS. I think it is because of the missing device tree overlay for the imx708 sensor. I also created a topic about this on [the forum](https://forums.balena.io/t/cant-get-libcamera-to-work-with-new-camera-module-v3/366977/2).

Jessseee, 2023-03-09:
I created a pull-request on meta-raspberrypi as that seems to be the correct place to add this overlay. https://github.com/agherzan/meta-raspberrypi/pull/1158

floion, 2023-04-29:
@Jessseee I see it's been merged in meta-raspberrypi but only in master. It would be good if it was merged in kirkstone also since that's what we are using here

Jessseee, 2023-11-12:
A backport of these changes is on its way for kirkstone, but it still needs to be reviewed. https://github.com/agherzan/meta-raspberrypi/pull/1197