balena-os/balena-raspberrypi #1009: Upgraded firmware

HakanL, 2023-04-04:
The latest BalenaOS build for Raspberry Pi CM4 (io board) has a firmware from January 2022. I would like to have an updated version of the firmware (these are files stored in the boot partition), and of course I can copy those manually to the boot folder, but I would like to understand how these can be updated in the build process (not automatically, but by re-pointing something, as I assume they get pulled in at some point during the build process). Best would be to a later version of the firmware be included in the official BalenaOS build for CM4.

floion, 2023-04-29:
Hi @HakanL this is how it was pinned to the Jan 2022 version: https://github.com/balena-os/balena-raspberrypi/commit/5af0a4f2d33c3f735b04eb860510c400a7ec6c4f

In the meanwhile we removed it because the BSP adds a newer version as the git commit I pointed to says. But it's an indication on how to update to a newer version by using the build system

floion, 2023-04-29:
So right now we include version 20220830.
Can this issue be closed now or do you need additional info?

HakanL, 2023-05-01:
Excellent, thank you for the information!

closed 2023-05-01 by HakanL