balena-os/balena-raspberrypi #1012: Build error - boot.scr

HakanL, 2023-05-02:
I'm getting a build error that boot.scr is missing, but when I look in the referenced directory the file is there.

Last part of the log:
```
NOTE: Running task 4495 of 4519 (virtual:native:/home/build/balena-raspberrypi/build/../layers/poky/meta/recipes-devtools/mtools/mtools_4.0.38.bb:do_populate_sysroot)
NOTE: recipe mtools-native-4.0.38-r0: task do_populate_sysroot: Started
NOTE: recipe mtools-native-4.0.38-r0: task do_populate_sysroot: Succeeded
NOTE: Running task 4496 of 4519 (virtual:native:/home/build/balena-raspberrypi/build/../layers/poky/meta/recipes-devtools/dosfstools/dosfstools_4.2.bb:do_populate_sysroot)
NOTE: recipe dosfstools-native-4.2-r0: task do_populate_sysroot: Started
NOTE: recipe dosfstools-native-4.2-r0: task do_populate_sysroot: Succeeded
NOTE: Running task 4497 of 4519 (/home/build/balena-raspberrypi/build/../layers/meta-raspberrypi/recipes-bsp/bootfiles/rpi-cmdline.bb:do_deploy)
NOTE: recipe rpi-cmdline-1.0-r0: task do_deploy: Started
NOTE: recipe rpi-cmdline-1.0-r0: task do_deploy: Succeeded
NOTE: Running task 4498 of 4519 (/home/build/balena-raspberrypi/build/../layers/meta-raspberrypi/recipes-bsp/bootfiles/rpi-config_git.bb:do_deploy)
NOTE: recipe rpi-config-git-r5: task do_deploy: Started
NOTE: recipe rpi-config-git-r5: task do_deploy: Succeeded
NOTE: Running task 4499 of 4519 (/home/build/balena-raspberrypi/build/../layers/meta-balena/meta-balena-common/recipes-core/images/balena-image.bb:do_resin_boot_dirgen_and_deploy)
NOTE: recipe balena-image-1.0-r0: task do_resin_boot_dirgen_and_deploy: Started
ERROR: balena-image-1.0-r0 do_resin_boot_dirgen_and_deploy: /home/build/balena-raspberrypi/build/tmp/deploy/images/raspberrypicm4-ioboard/boot.scr is an invalid path referenced in BALENA_BOOT_PARTITION_FILES.
ERROR: balena-image-1.0-r0 do_resin_boot_dirgen_and_deploy: ExecutionError('/home/build/balena-raspberrypi/build/tmp/work/raspberrypicm4_ioboard-poky-linux/balena-image/1.0-r0/temp/run.do_resin_boot_dirgen_and_deploy.624913', 1, None, None)
ERROR: Logfile of failure stored in: /home/build/balena-raspberrypi/build/tmp/work/raspberrypicm4_ioboard-poky-linux/balena-image/1.0-r0/temp/log.do_resin_boot_dirgen_and_deploy.624913
NOTE: recipe balena-image-1.0-r0: task do_resin_boot_dirgen_and_deploy: Failed
ERROR: Task (/home/build/balena-raspberrypi/build/../layers/meta-balena/meta-balena-common/recipes-core/images/balena-image.bb:do_resin_boot_dirgen_and_deploy) failed with exit code '1'
NOTE: Running task 4500 of 4519 (/home/build/balena-raspberrypi/build/../layers/meta-raspberrypi/recipes-bsp/rpi-u-boot-scr/rpi-u-boot-scr.bb:do_deploy)
NOTE: recipe rpi-u-boot-scr-1.0-r0: task do_deploy: Started
NOTE: recipe rpi-u-boot-scr-1.0-r0: task do_deploy: Succeeded
NOTE: recipe parted-native-3.4-r0: task do_compile: Succeeded
NOTE: Tasks Summary: Attempted 4500 tasks of which 3871 didn't need to be rerun and 1 failed.
```

Error: **ERROR: balena-image-1.0-r0 do_resin_boot_dirgen_and_deploy: /home/build/balena-raspberrypi/build/tmp/deploy/images/raspberrypicm4-ioboard/boot.scr is an invalid path referenced in BALENA_BOOT_PARTITION_FILES.**

`boot.scr` is in this folder: `/home/build/balena-raspberrypi/build/tmp/deploy/images/raspberrypicm4-ioboard/`, 516 bytes.

HakanL, 2023-05-02:
If I run it again then it will continue where it left off and be successful.

floion, 2023-05-03:
Has this happened on a new clone and build or when building in an older existing clone which had been built before?
Also, are you using the shared sstate option when building?

HakanL, 2023-05-03:
@floion I was struggling to find documentation for how to build the images from scratch, it seems there were version-specific dependencies on OS version, docker, etc. I was hoping to find a Docker environment that would do the full builds, but was unable to, so I ended up writing that (see here: https://forums.balena.io/t/new-linux-kernel-build-for-raspberry-pi-cm4/366630) and the repo https://github.com/DMXCore/BalenaOS-Build, which is what I'm using to build (but getting this error). It's essentially a new clone (but I have updated it instead of complete clone over the last 3 months). I'm not aware of the shared state option, maybe I'm building it wrong, I'd love to set it up better, in fact, to set it up as a Github action would be the ultimate goal.