balena-os/balena-raspberrypi #1085: RS-485 not working with revpi-connect-4 (new device)

wickywaka, 2023-12-13:
We added support for revpi-connect-4 here https://github.com/balena-os/balena-raspberrypi/pull/1078

During further testing, I see that we don't have a device node/file in /dev for rs-485. So may be there is something missing in the device trees. Any idea what can it be?

ddtdanilo, 2024-03-18:
any news on this?

ludovic-cook-e, 2024-05-23:
Hi, I'm not a Balena OS user, but I also had to create an image for the RevPi Connect 4.

I had the RS485 working in my image by adding this udev file (https://gitlab.com/revolutionpi/revpi-base-files/-/blob/v0.2.0/udev/50-revpi.rules) into `${D}${sysconfdir}/udev/rules.d/` with the following patch 

```
--- a/50-revpi.rules
+++ b/50-revpi.rules
@@ -2,7 +2,7 @@
 # 
 # SPDX-License-Identifier: GPL-2.0-or-later
 
-PROGRAM="/bin/grep -Fz -m 1 'kunbus,revpi-' /sys/firmware/devicetree/base/compatible"
+PROGRAM="/bin/grep -F -m 1 'kunbus,revpi-' /sys/firmware/devicetree/base/compatible"
 RESULT=="kunbus,revpi-core", GOTO="revpi_core"
 RESULT=="kunbus,revpi-core-2022", GOTO="revpi_core"
 RESULT=="kunbus,revpi-core-s-2022", GOTO="revpi_core"
```

Doing so seems to create `/dev/ttyRS485`, which is a working RS485 interface.

Btw, I don't have `/dev/node/file` in my image, so there might be more work to be done to link the RS485 to this interface.

I hope this helps :)

floion, 2025-07-22:
Should be fixed by https://github.com/balena-os/balena-raspberrypi/pull/1263