balena-os/balena-raspberrypi #1098: linux-firmware: Include MT7610U and MT7612U firmware

aolson-lg, 2024-01-31:
I have recently tested Panda PAU0B and PAU0D adapters on Raspberry Pi 4B running Balena OS. I can see that the kernel loads the modules for these two Panda adapters but it fails to load the firmware for MT7610U and MT7612U. Is there anyway add these firmware to he relevant device repository?

Balena OS version
root@3d54bd3:/mnt/boot/system-connections# hostnamectl
Static hostname: 3d54bd3
Icon name: computer
Machine ID: b5350b0c625c4dffb98089977585f430
Boot ID: ab302a4839174d0fb63c87c94161ed37
Operating System: balenaOS 4.1.3+rev2
Kernel: Linux 5.15.92-v8
Architecture: arm64

floion, 2024-02-29:
Hi, right now the rootfs space is running out so we need to be careful with adding new files. This should be solved when we have the migrator in place

aolson-lg, 2024-03-01:
Hello, Is there any workaround for now so we can test these adapter on our Raspberry Pi 4B?